riscv-boom/riscv-boom #292: SBT warns me that Chisel dependency cannot be resolved

novakboskov, 2019-06-11:
I cloned the repository as is described in your [Directions](https://github.com/riscv-boom/riscv-boom#directions) then run `sbt` in the root directory. It warned me about possible incompatibilities in my libraries. Then I run `evicted` and got that
```
[warn] 	module not found: edu.berkeley.cs#chisel3_2.11;3.2-SNAPSHOT
[warn] ==== local: tried
[warn]   /home/myname/.ivy2/local/edu.berkeley.cs/chisel3_2.11/3.2-SNAPSHOT/ivys/ivy.xml
[warn] ==== public: tried
[warn]   https://repo1.maven.org/maven2/edu/berkeley/cs/chisel3_2.11/3.2-SNAPSHOT/chisel3_2.11-3.2-SNAPSHOT.pom
[warn] ==== local-preloaded-ivy: tried
[warn]   /home/myname/.sbt/preloaded/edu.berkeley.cs/chisel3_2.11/3.2-SNAPSHOT/ivys/ivy.xml
[warn] ==== local-preloaded: tried
[warn]   file:////home/myname/.sbt/preloaded/edu/berkeley/cs/chisel3_2.11/3.2-SNAPSHOT/chisel3_2.11-3.2-SNAPSHOT.pom
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::          UNRESOLVED DEPENDENCIES         ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: edu.berkeley.cs#chisel3_2.11;3.2-SNAPSHOT: not found
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn]
[warn] 	Note: Unresolved dependencies path:
[warn] 		edu.berkeley.cs:chisel3_2.11:3.2-SNAPSHOT
[warn] 		  +- edu.berkeley.cs:hardfloat_2.11:1.2
[error] sbt.librarymanagement.ResolveException: unresolved dependency: edu.berkeley.cs#chisel3_2.11;3.2-SNAPSHOT: not found
[error] 	at sbt.internal.librarymanagement.IvyActions$.resolveAndRetrieve(IvyActions.scala:332)
```
However, I have checked Maven and `edu.berkeley.cs:chisel3_2.11:3.1.7` is there but not `3.2-SNAPSHOT`. 

Why does it look for `edu.berkeley.cs:chisel3_2.11:3.2-SNAPSHOT`? Where can I change it?

I am not familiar with the ecosystem but it looks like something on `rocket-chip` side (`boom-template/rocket-chip/build.sbt`):
```
lazy val chisel = (project in file("chisel3")).settings(commonSettings)
lazy val hardfloat  = project.dependsOn(chisel).settings(commonSettings)
  .settings(crossScalaVersions := Seq("2.11.12", "2.12.4"))
lazy val macros = (project in file("macros")).settings(commonSettings)
lazy val rocketchip = (project in file("."))
  .settings(commonSettings, chipSettings)
  .dependsOn(chisel, hardfloat, macros)
  .aggregate(chisel, hardfloat, macros) // <-- means the running task on rocketchip is also run by aggregate tasks
```

abejgonzalez, 2019-06-13:
Hi. All the developers here at Berkeley working on BOOM don't directly call `sbt` from the home directory to build RTL. Instead, we wrap our `sbt` calls with the `Makefile` found in the `verisim` and `vsim` directories (hence the `make CONFIG=BoomConfig` in the instructions). I just recloned the repo and invoked `make` and had no problems. 

Additionally, we use the `sbt` jar given by Rocket Chip instead of using one provided by the host OS. See https://github.com/riscv-boom/boom-template/blob/8241911d3fa13ab81df276899c2ab839fd8b3912/Makefrag-variables#L49.

Here at Berkeley, the team is looking to migrate from `boom-template` to a new repository that should make building RTL, and doing other stuff in the ecosystem a bit easier to manage. If you are willing to try the "alpha" of sorts, it is located at https://github.com/ucb-bar/project-template. It has some docs associated with it that might be good to read also (https://bar-project-template.readthedocs.io/). The developers here at Berkeley work off the `rebar-dev` branch of this repo (has the most up-to-date things but can be buggy). However, the `master` branch should be stable.

With that being said... what exactly are you trying to do?

sequencer, 2019-06-13:
You may need to `sbt publishLocal` from chisel3 project, because chisel3.2 is not ready for publish to public Maven currently, you need to package your own chisrl3.2 as SNAPSHOT.

closed 2021-05-01 by jerryz123