riscv-boom/riscv-boom #363: DCache ECC

siyy123, 2019-07-31:
the compiler crashed when turing the .fir into .v .the config is MegaBoomECCConfig. Here is the log:

java -Xmx8G -Xss8M -cp "/home/siyy123/riscv/boom-template/rocket-chip/firrtl/utils/bin/firrtl.jar":""/home/siyy123/riscv/boom-template/rocket-chip/target/scala-2.12/classes:/home/siyy123/riscv/boom-template/rocket-chip/chisel3/target/scala-2.12/*"" firrtl.Driver -i /home/siyy123/riscv/boom-template/vsim/generated-src/boom.system.TestHarness.MegaBoomECCConfig.fir -o /home/siyy123/riscv/boom-template/vsim/generated-src/boom.system.TestHarness.MegaBoomECCConfig.v -X verilog -faf /home/siyy123/riscv/boom-template/vsim/generated-src/boom.system.TestHarness.MegaBoomECCConfig.anno.json --infer-rw TestHarness --repl-seq-mem -c:TestHarness:-o:/home/siyy123/riscv/boom-template/vsim/generated-src/boom.system.TestHarness.MegaBoomECCConfig.conf
Exception in thread "main" firrtl.passes.CheckWidths$BitsWidthException:  @[dcache.scala 280:52:boom.system.TestHarness.MegaBoomECCConfig.fir@131783.4]: [target ~TestHarness|BoomNonBlockingDCache] High bit 129 in bits operator is larger than input width 128 in bits(tl_out.d.bits.data, 129, 0).
        at firrtl.passes.CheckWidths$.check_width_e$1(CheckWidths.scala:81)
        at firrtl.passes.CheckWidths$.$anonfun$run$8(CheckWidths.scala:98)
        at firrtl.passes.CheckWidths$.$anonfun$run$8$adapted(CheckWidths.scala:98)
        at firrtl.ir.DefNode.foreachExpr(IR.scala:330)
        at firrtl.traversals.Foreachers$StmtForMagnet$$anon$2.foreach(Foreachers.scala:21)
        at firrtl.traversals.Foreachers$StmtForeach$.foreach$extension(Foreachers.scala:35)
        at firrtl.passes.CheckWidths$.check_width_s$1(CheckWidths.scala:98)
        at firrtl.passes.CheckWidths$.$anonfun$run$10(CheckWidths.scala:99)
        at firrtl.passes.CheckWidths$.$anonfun$run$10$adapted(CheckWidths.scala:99)
        at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:59)
        at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:52)
        at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
        at firrtl.ir.Block.foreachStmt(IR.scala:363)
        at firrtl.traversals.Foreachers$StmtForMagnet$$anon$1.foreach(Foreachers.scala:18)
        at firrtl.traversals.Foreachers$StmtForeach$.foreach$extension(Foreachers.scala:35)
        at firrtl.passes.CheckWidths$.check_width_s$1(CheckWidths.scala:99)
        at firrtl.passes.CheckWidths$.$anonfun$run$17(CheckWidths.scala:121)
        at firrtl.passes.CheckWidths$.$anonfun$run$17$adapted(CheckWidths.scala:121)
        at firrtl.ir.Module.foreachStmt(IR.scala:682)
        at firrtl.traversals.Foreachers$ModuleForMagnet$$anon$11.foreach(Foreachers.scala:79)
        at firrtl.traversals.Foreachers$ModuleForeach$.foreach$extension(Foreachers.scala:92)
        at firrtl.passes.CheckWidths$.check_width_m$1(CheckWidths.scala:121)
        at firrtl.passes.CheckWidths$.$anonfun$run$19(CheckWidths.scala:124)
        at firrtl.passes.CheckWidths$.$anonfun$run$19$adapted(CheckWidths.scala:124)
        at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:59)
        at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:52)
        at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
        at firrtl.passes.CheckWidths$.run(CheckWidths.scala:124)
        at firrtl.passes.Pass.execute(Passes.scala:24)
        at firrtl.passes.Pass.execute$(Passes.scala:22)
        at firrtl.passes.CheckWidths$.execute(CheckWidths.scala:12)
        at firrtl.Transform.$anonfun$runTransform$2(Compiler.scala:199)
        at firrtl.Utils$.time(Utils.scala:186)
        at firrtl.Transform.runTransform(Compiler.scala:199)
        at firrtl.SeqTransformBased.$anonfun$runTransforms$1(Compiler.scala:264)
        at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:122)
        at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:118)
        at scala.collection.immutable.List.foldLeft(List.scala:86)
        at firrtl.SeqTransformBased.runTransforms(Compiler.scala:264)
        at firrtl.SeqTransformBased.runTransforms$(Compiler.scala:263)
        at firrtl.SeqTransform.runTransforms(Compiler.scala:268)
        at firrtl.SeqTransform.execute(Compiler.scala:274)
        at firrtl.Transform.$anonfun$runTransform$2(Compiler.scala:199)
        at firrtl.Utils$.time(Utils.scala:186)
        at firrtl.Transform.runTransform(Compiler.scala:199)
        at firrtl.Compiler.$anonfun$compile$2(Compiler.scala:443)
        at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:122)
        at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:118)
        at scala.collection.immutable.List.foldLeft(List.scala:86)
        at firrtl.Compiler.$anonfun$compile$1(Compiler.scala:443)
        at firrtl.Utils$.time(Utils.scala:186)
        at firrtl.Compiler.compile(Compiler.scala:443)
        at firrtl.Compiler.compile$(Compiler.scala:439)
        at firrtl.VerilogCompiler.compile(LoweringCompilers.scala:162)
        at firrtl.Driver$.$anonfun$execute$1(Driver.scala:242)
        at logger.Logger$.$anonfun$makeScope$2(Logger.scala:138)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
        at logger.Logger$.makeScope(Logger.scala:136)
        at firrtl.Driver$.execute(Driver.scala:224)
        at firrtl.Driver$.execute(Driver.scala:302)
        at firrtl.Driver$.main(Driver.scala:318)
        at firrtl.Driver.main(Driver.scala)
/home/siyy123/riscv/boom-template/Makefrag:37: recipe for target '/home/siyy123/riscv/boom-template/vsim/generated-src/boom.system.TestHarness.MegaBoomECCConfig.v' failed
make: *** [/home/siyy123/riscv/boom-template/vsim/generated-src/boom.system.TestHarness.MegaBoomECCConfig.v] Error 1

It seems that the DCache has some bug when ECC is enabled. And there is no error if ECC is disabled. Could someone fix it?

jerryz123, 2019-07-31:
This is actually a rocketchip error, since this version of BOOM uses the rocketchip NBDCache, which is where ECC is implemented.

The fix is probably to just not use the MegaBoomECCConfig, or modify it to not add DCache ECC.

jerryz123, 2019-07-31:
I just found https://github.com/chipsalliance/rocket-chip/issues/495 which seems to indicate NBDCache ECC is known to be not functional.

ccelio, 2019-07-31:
It's not enough for the NBDCache to support ECC (which is does not anyways); you would also need to add support in the BOOM core/LSU to support handling correctable and uncorrectable errors. For example, if a store fails with an uncorrectable error, how do you handle this? It will not be precise, but you probably want to throw an exception and crash out for the SW to handle.

jerryz123, 2021-05-01:
ECC is no longer supported, since we use fully custom L1 caches

closed 2021-05-01 by jerryz123