riscv-boom/riscv-boom #412: Aliasing in BTB for branch instructions.

hyf6661669, 2019-11-17:
I just started to study riscv-boom several days ago, if I'm wrong please correct me. BTB only stores partial tags for instructions, so aliasing could happen for branch instructions. But when it happens, it seems that you don't correct it in f3-stage. 

For example, if we have 2 fetch packets:
Fetch boundary of 1st fetch packet is 0x0000_0200, it contains 4 RVC instructions:
0x0000_0200: rvc_0
0x0000_0202: rvc_1
0x0000_0204: c.beqz to target_0
0x0000_0206: rvc_2
Fetch boundary of 2nd fetch packet is 0x8000_0200, it also contains 4 RVC instructions:
0x8000_0200: rvc_3
0x8000_0202: rvc_4
0x8000_0204: c.beqz to target_1
0x8000_0206: rvc_5

Let's assume that btb has already stored information about 1st fetch packet and will predict instruction at 0x0000_0204 as taken. But btb hasn't recorded information about 2nd fetch packet. If we begin to execute instructions in 2nd fetch packet. It will point to the same entry in btb (because their partial tags are equal and aliasing happens).  Therefore, when fetch_pc = 0x8000_0200, btb will also predict that frontend should jump to target_0.

In f3-stage, the complex branch predictor(gshare or tage) maybe also predict ```c.beqz to target_1``` as taken. 

In fetch-control-unit.scala:
https://github.com/riscv-boom/riscv-boom/blob/master/src/main/scala/ifu/fetch-control-unit.scala#L365-L396
```
  when (f3_valid && f3_btb_resp.valid) {
    // btb made a prediction
    // Make a redirect request if:
    //    - the BPD (br) comes earlier than the BTB's redirection.
    //    - If both the BTB and the BPD predicted a branch, the BPD wins (if disagree).
    //       * involves refetching the next cacheline and undoing the current packet's mask if we "undo" the BT's
    //       taken branch.

    val btb_idx = f3_btb_resp.bits.cfi_idx

    when (BpredType.isAlwaysTaken(f3_btb_resp.bits.bpd_type)) {
      f3_bpd_may_redirect_taken := io.f3_bpd_resp.valid && f3_bpd_br_taken && f3_bpd_br_idx < btb_idx

      assert (f3_btb_resp.bits.taken)
    } .elsewhen (f3_btb_resp.bits.taken) {
      // does the bpd predict the branch is taken too? (assuming bpd_valid)
      val bpd_agrees_with_btb = f3_bpd_predictions(btb_idx)
      f3_bpd_may_redirect_taken := io.f3_bpd_resp.valid && f3_bpd_br_taken &&
        (f3_bpd_br_idx < btb_idx || !bpd_agrees_with_btb)
      f3_bpd_may_redirect_next := io.f3_bpd_resp.valid && !f3_bpd_br_taken

      assert (BpredType.isBranch(f3_btb_resp.bits.bpd_type))
    } .elsewhen (!f3_btb_resp.bits.taken) {
      f3_bpd_may_redirect_taken := io.f3_bpd_resp.valid && f3_bpd_br_taken
    }
  } .otherwise {
    // BTB made no prediction - let the BPD do what it wants
    f3_bpd_may_redirect_taken := io.f3_bpd_resp.valid && f3_bpd_br_taken
    // add branch to the BTB if we think it will be taken
    f3_bpd_btb_update_valid := f3_bpd_may_redirect_taken
  }

```
We have f3_bpd_may_redirect_taken = f3_bpd_may_redirect_next  = 0.

Although in branch-checker.scala:
https://github.com/riscv-boom/riscv-boom/blob/master/src/main/scala/ifu/branch-checker.scala#L77-L105
```
  when (io.btb_resp.valid) {
    when (io.btb_resp.bits.cfi_type === CFI_BR && (io.btb_resp.bits.taken || bpd_predicted_taken)) {
      wrong_cfi := !io.is_br(btb_idx)
      wrong_target := io.br_targs(btb_idx) =/= btb_target
    } .elsewhen (io.btb_resp.bits.cfi_type === CFI_JAL) {
      wrong_cfi := !io.is_jal(btb_idx)
      wrong_target := io.jal_targs(btb_idx) =/= btb_target
    } .elsewhen (io.btb_resp.bits.cfi_type === CFI_JALR) {
      wrong_cfi := !io.is_jr(btb_idx)
    } .otherwise {
      wrong_cfi := io.btb_resp.bits.cfi_type === CFI_X && io.btb_resp.bits.taken
      when (io.valid) {
        assert (io.btb_resp.bits.cfi_type =/= CFI_X, "[fetch] predicted on a non-cfi type.")
      }
    }
  }

  val nextline_pc = nextFetchStart(io.aligned_pc)

  val btb_was_wrong = io.btb_resp.valid && (wrong_cfi || wrong_target || !io.inst_mask(btb_idx))

  val jal_idx = PriorityEncoder(io.is_jal.asUInt)
  val btb_hit  = io.btb_resp.valid
  val jal_wins = io.is_jal.reduce(_|_) &&
                 (!btb_hit ||
                 btb_was_wrong ||
                 (jal_idx < btb_idx) ||
                 !io.btb_resp.bits.taken)
```

We would find that wrong_target = 1 and btb_was_wrong = 1, it will just redirect f0-stage to read the next fetch packet.

jerryz123, 2019-11-17:
If I understand correctly, your observation is that the BTBs will cause a incorrect redirection when the the frontend sees a new branch that is aliased with another branch stored in the BTB. Do you see this as a frequent occurence? This would only be a problem if the code ping-pongs between two branches that alias each other in both partial tag and index, which seems like a rare situation to me.

hyf6661669, 2019-11-18:
> 
> 
> If I understand correctly, your observation is that the BTBs will cause a incorrect redirection when the frontend sees a new branch that is aliased with another branch stored in the BTB. Do you see this as a frequent occurence? This would only be a problem if the code ping-pongs between two branches that alias each other in both partial tag and index, which seems like a rare situation to me.

Sorry, I am just reading the code without doing any simulation. I'm still studying how to use chipyard..

ccelio, 2019-11-18:
> Sorry, I am just reading the code without doing any simulation. I'm still studying how to use chipyard..

Reading code is a great start to catching small issues. Once you have chipyard running, I recommend you add a performance event to capture this scenario you are concerned about, and then measure how often it happens for applications you care about.

jerryz123, 2021-05-01:
New branch predictors fixed this.

closed 2021-05-01 by jerryz123