riscv-boom/riscv-boom #416: Add commit-log printf DPI widget.

jerryz123, 2019-11-29:
**Type of issue**: bug report | feature request | question | other enhancement

**Impact**: no rtl change | rtl refactoring | new rtl | unknown

**Development Phase**: request | proposal

**Other information**

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

abejgonzalez, 2019-12-05:
Thinking about this a bit more, Im not exactly sure how you would pass the formatted string into the Verilog module (since it can vary cycle by cycle). But I agree this would be very nice.

I think a simple solution without a lot of change would be to have something like:
```
val commitUtil = new PrintUtil("[COMMIT]")
commitUtil.printf(%d... format string", arg1, arg2, ...)
```

where the output looks like
```
[COMMIT] STRING TO PRINT HERE
```

This would make it very easy to filter out certain prints from others using a simple `grep`.

jerryz123, 2019-12-05:
The point of this is to avoid post-processing with grep or sed.

You can pass a string into a blackbox module through the IO, as a fixed-length array of bytes. (We would have to mandate that the printf strings have a length limit). The tricky thing is to have the blackboxes figure out what file to write the strings to.