riscv-boom/riscv-boom #428: No repair mechanism for return address stack (RAS)

hyf6661669, 2019-12-19:
Hi, I find that there is no repair mechanism for RAS in riscv-boom, there are many reasons that will cause a pipeline flush:
1. jalr/conditional branch mispredictions
2. misspeculation load instructions
3. instructions which are related to the core/ISA architecture, like fence/fence.i/sfence/csr

The instructions (or fetch packets) behind these instructions may change the contents of RAS before the pipeline flush happens. But after the pipeline is flushed, the latter "refetched" instructions will see the modified RAS, instead of the original RAS. Under this situation, the following CALL/RET instructions are very likely to be mispredicted. However, use checkpoints for the whole RAS could be very expensive. Is there any cost-effective methods for repairing RAS?

ccelio, 2019-12-19:
This is a very good observation.

Unfortunately, at this time, it is left as an exercise to the reader to provide pull requests that improve the situation. =) I've not searched the published literature myself, but perhaps "return address stack repair" might bring up some hints for you to investigate.

jerryz123, 2019-12-19:
This will be addressed in a upcoming version of BOOM. I have a version of the RAS that repairs the stack index and the top-of-stack address

closed 2019-12-20 by hyf6661669