riscv-boom/riscv-boom #436: When to use double parentheses?

ppoocde, 2020-02-06:
Hello all,

I am learning chisel and read Boom code as reference. 
I came across this class definition:
class LSUExeIO(implicit p: Parameters) extends BoomBundle()(p)
{
  // The "resp" of the maddrcalc is really a "req" to the LSU
  val req       = Flipped(new ValidIO(new FuncUnitResp(xLen)))
  // Send load data to regfiles
  val iresp    = new DecoupledIO(new boom.exu.ExeUnitResp(xLen))
  val fresp    = new DecoupledIO(new boom.exu.ExeUnitResp(xLen+1)) // TODO: Should this be fLen?
}

this class definition extends BoomBundle, and put ()(p) after.

The definition of BoomBundle is :
class BoomBundle(implicit val p: Parameters) extends freechips.rocketchip.util.ParameterizedBundle  with HasBoomCoreParameters

p is defined as implicit in BoomBundle. it is the only parameter. My question is why the code put two parentheses after extends BoomBundle?

Thanks,

Sunny

**Type of issue**: question

abejgonzalez, 2020-02-08:
Hi,

I'm not a Scala power user by any means but this is my understanding:

If you use `implicit` in the argument list, all parameters in that `()` are considered `implicit` and thus the compiler will try to fill in that variable for you.

Take this:
```
class C(x: Int, y: Int)(implicit z: Int){...}
```

Here the `z` parameter is the only `implicit` parameter. If you move the `implicit z` into the prior parameter list, the compile will break.

To instantiate the `C` class, I can do it a couple of ways:

```
val c0 = new C(a,b) // z is automatically passed in
val c1 = new C(a,b)(z) // z is explicitly passed in
```

Now why does `BoomBundle` have the `()(p)`, I think it is mainly to keep things consistent... i.e. if `BoomBundle` had some non-`implicit` parameters, you could put them in the empty `()` then put the `(p)` afterwards. As you can see in the other parts of the code we are trying to clean this up so that the code is cleaner (https://github.com/riscv-boom/riscv-boom/blob/6bb61d2a97810542a44fbf3aacc02c9f2c03ad27/src/main/scala/exu/register-read/regfile.scala#L30).

Hopefully, that clarifies things a bit.

ppoocde, 2020-02-10:
thank you for your help i really appreciate it

closed 2020-02-10 by ppoocde