riscv-boom/riscv-boom #439: Why was TAGE branch predictor disabled?

jfustos, 2020-02-19:
**Type of issue**: bug report | feature request | question | other enhancement
question

**Impact**: no rtl change | rtl refactoring | new rtl | unknown

**Development Phase**: request | proposal

**Other information**

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

Hello, I am wondering the reason that the TAGE predictors were disabled? I see a comment when they were disabled that just says that they are being disabled for now and an issue that says that they were not predicting as well as expected at some point. Was this done because of some bug that couldn't be found, the implementation just took up too much space for the benefit, or some other reason? I am currently going to spend some time trying to improve the branch prediction for a project that I am working on and want to make sure am not missing important information from the past.

Thanks.

jerryz123, 2020-02-19:
TAGE was disabled because the existing implementation had known performance issues, and was very difficult to debug. The existing TAGE could predict only 1 instruction per cycle, used path history instead of branch history, and was frequently corrupted by misspeculation

A working implementation of TAGE that resolves these issues will be released soon.

jfustos, 2020-02-19:
Thanks for the fast response @jerryz123!!!

closed 2020-02-19 by jfustos