riscv-boom/riscv-boom #454: Behavior mismatch between Spike and Boom

JaewonHur, 2020-04-17:
**Type of issue**: bug report

**Impact**:  unknown

**Other information**

When I run fle.d instruction in spike and boom,
Spike does not set FS bits in mstatus if both operands are not NaN.
However, Boom set FS bits to dirty.

I'm using Boom commit: a26504f34db8327a5d4d9bb3a342dddf79e02b35
and Spike commit: 157143b5038ef2b92b1c70d445b566d57b6f5096 
Could you please check this behavior?

Here is all my log and test files.

[test.zip](https://github.com/ucb-bar/chipyard/files/4491270/test.zip)

jerryz123, 2020-04-17:
Hi Jaewon.

Thanks for the very detailed bug report.

This divergence from Spike is an issue, as it makes verification a bit harder.
However, BOOM's conservative handling of MSTATUS.FS is permitted by the RISC-V privileged SPEC. Specifically, these lines in section 3.1.6.5: 

> Implementations may choose to track the dirtiness of the floating-point register state imprecisely by reporting the state to be dirty even when it has not been modified.  On some implementations, **some  instructions  that  do  not  mutate  the  floating-point  state  may  cause  the  state  to  transition from Initial or Clean to Dirty.**  On other implementations, dirtiness might not be tracked at all,in which case the valid FS states are Off and Dirty, and an attempt to set FS to Initial or Clean causes it to be set to Dirty

Currently, BOOM conservatively dirties FS for all floating point instructions, instead of only those which write to a floating point register. I will try to fix this soon, but the current behavior should not cause errors in correctly written software.

JaewonHur, 2020-04-20:
Thanks for confirm.