riscv-boom/riscv-boom #457: different result of sc.d after exception.

JaewonHur, 2020-04-22:
**Type of issue**: bug report 

**Impact**: unknown

**Development Phase**: proposal

Hi again.
Recently, I'm comparing boom with spike and I find a mismatch when there is an exception between lr.d and sc.d.

As in https://github.com/riscv/riscv-isa-sim/issues/450#issuecomment-617400248,
RISC-V does not guarantee sc.d to succeed if there is an exception.
(sc.d fails in spike)

But when running the program in Boom,
Boom has different result depending on the executed instructions.
(When there is a `div` instruction like the first instruction sequence, sc.d fails
bug when `nop` instruction like second one, then sc.d succeed)

```
...
lr.d x3,(x4)
...
exception
...
div x27,x7,x14
...
sc.d x9, zero, (x4)
...
```

```
...
lr.d x3,(x4)
...
exception
...
nop
...
sc.d x9, zero, (x4)
...
```

I don't know this could be an issue, but at lease I think it is noteworthy to know this behavior.
Here's my test.

[test.zip](https://github.com/riscv-boom/riscv-boom/files/4513213/test.zip)

jerryz123, 2020-04-22:
Thanks for the report.

As you noted, this is not a correctness issue. There are two microarchitectural details which result in this divergence.

The first is BOOM's implementation of the LR/SC mechanism, which only needs to guarantee successes for short sequences of integer-only branch-free code.

>  The static code for the LR/SC sequence plus the code to retry the sequence in case of failure must comprise **at most 16 integer instructions** placed sequentially in memory. For the sequence to be guaranteed to eventually succeed, the dynamic code executed between the LR and SC 
 instructions can **only contain other instructions from the base “I” subset, excluding loads, stores, backward jumps or taken backward branches, FENCE, FENCE.I, and SYSTEM instructions.**

There is a small counter in the DCache which counts down cycles after a `lr`, after which it yields the reservation. For a `sc` to succeed, it must fire through the DCache before this counter expires. The constrained set of sequences which must succeed should always issue quickly enough that the `sc` arrives before the counter expires.

The second detail is that currently, BOOM waits for all older instructions to commit before issuing the `sc`.

In the first example sequence you provide, the `sc` must wait for the  long-latency `div` instruction to retire and commit, which delays its execution long enough that it misses the success window for the `lr`. In the second sequence, the `nop` instruction completes instantly and does not delay the `sc`, allowing it to succeed.

JaewonHur, 2020-04-22:
Ah, I see. 
Thanks for kind explanation.

closed 2020-04-22 by JaewonHur