riscv-boom/riscv-boom #458: Invalid rm in floating point instructions

JaewonHur, 2020-04-23:
**Type of issue**: question

**Impact**: unknown

**Development Phase**: request | proposal

**Other information**

Hi, when executing Boom core with floating point instruction which has invalid rm (like 5 or 6), I see Boom just commits the instruction.
(Spike raise illegal instruction exception)

Could you tell me how Boom handles such cases?

Thanks for the answers.

jerryz123, 2020-04-23:
Can you provide example program which demonstrates this?

JaewonHur, 2020-04-24:
[test.zip](https://github.com/riscv-boom/riscv-boom/files/4526432/test.zip)

like this?
It's somewhat awkward but it still works.

jerryz123, 2020-04-29:
Haha there's a nice TODO marking this hole in the implementation.
https://github.com/riscv-boom/riscv-boom/blob/e81d8ccbdb2906fc165fedaed8fb264d9f824037/src/main/scala/exu/decode.scala#L489

I will work on this soon. 

Thanks again, for finding these divergences.

JaewonHur, 2020-04-29:
Thanks for confirm.