riscv-boom/riscv-boom #459: source field in ProbeAckData

JaewonHur, 2020-04-29:
**Type of issue**: bug report 

**Impact**: unknown

When RocketTile with Boom core receives PROBE_REQUEST message through tilelink b ports, it responds with with PROBE_ACK (_DATA) messages.
The source fields of PROBE_REQUEST message and PROBE_ACK message should be same as in tilelink-spec-1.8.0, however they have different value in the simulation.

Could you check this behavior?

Here's the waveform and you can see those messages (tl_b, tl_c) near the last.

[waveform.zip](https://github.com/riscv-boom/riscv-boom/files/4550164/waveform.zip)

jerryz123, 2020-04-29:
Thanks for finding this. Can you see if this fix works for you? The fix should be to change this line
https://github.com/riscv-boom/riscv-boom/blob/e81d8ccbdb2906fc165fedaed8fb264d9f824037/src/main/scala/lsu/dcache.scala#L66

to
```
fromSource = req.source,
```

In the original DCache design (from Rocket), voluntary writebacks are always executed before a refill request, so the source id for the `Release` (voluntary eviction) message is reused from the source id of the MSHR which will handle the `Refill` once the `Release` is complete. In the new BOOM DCache, to support parallel `Refill` and `Release`. the `WritebackUnit` is allocated its own source id to use.
However, the `WritebackUnit` should not use this source-id for `ProbeAck`, as you describe. Instead it should use the source-id of the `Probe`

jerryz123, 2020-04-29:
I'm very curious why this didn't cause any issues with my testing... I've tested this with TLMonitors, which *should* verify TileLink conformance.

BOOM also runs large programs successfully with this bug present, so the other components in the memory hierarchy must be tolerant to this bug?

JaewonHur, 2020-04-29:
Yes, your patch fixes the bug.
Now they have same source id.

Maybe other components in the hierarchy does not check the source id as seriously, but I couldn't test with the other components because I'm only testing on RocketTile which contains BoomCore.

ddxoxpp, 2021-11-05:
Broadcaster doesn't know each cache's each sourceID's client function in the sourceID range of each cache's, so it simply send probe message with each cache‘s first sourceID. But in Boom's dcache ,the first sourceID is for mshr, while the last sourceID in sourceID range is for probeack(data) message. MSHR in Boom don't use B,C channel, so we can use the MSHR's sourceID for probe and probeack(data) message ?

jerryz123, 2021-11-07:
According to TileLink spec the Writeback response to a Probe must carry the same sourceId that the Probe was originally sent with. In BOOM the writeback unit did not respect this, which caused some TileLink compliance assertions to fire. The patch I posted above fixes the issue.