riscv-boom/riscv-boom #469: BOOM compiled by dc report only 250M frequency

qqjinger, 2020-06-16:
**BOOM version：**https://github.com/riscv-boom/riscv-boom/tree/779c62c5634847b517be64c554af66829de40067
**Process：**SMIC 40nm
**Detail description：**
      Using design compiler of synopsys，I only get 250M frequency for the above mentioned version BOOM. 
     I don't know whether BOOM version, or the configuration when generating BOOM verilog, or the fundary process result into low frequency or not. Is there anyone synthesize BOOM? And getting what frequency?
      Thank you!

jerryz123, 2020-06-16:
I can achieve 1 GHz frequency with a commercial 22nm process. What configuration are you using, and what critical paths do you see?

qqjinger, 2020-06-17:
BOOM configuration:  L1 32KB，L2 256KB，double core，no accelerator，BOOM V2. 
Critical paths：the critical paths is as follow:
     Startpoint: u_core/u_boom_top/l2/mods_0/abc_mshrs_5/s_execute_reg
     Endpoint: u_core/u_boom_top/l2/mods_0/c_mshr/s_writeback_reg
     Critical Path Length:          2.80ns (355Mhz)
Attention: I did not limit area, so dc reported a better result, but the frequency is still not ideal.
Which BOOM version are you  using? Thanks!

jerryz123, 2020-06-22:
I use version 3. Its strange to me that your path is in the L2 at such a low frequency.

qqjinger, 2020-06-23:
After adjusting dc constraint scripts, for boom v2 in smic 40nm，I can reach up to 417MHz ideally. And for boom v3，I can get 580Mhz for small configs in the same process.
    So, Whether the frequency is related to L2 configuration size？Can you share your L1+L2 config info?Or can small configs get higher frequency?

jerryz123, 2020-08-11:
I use the default L1 config. If the L2 is on the critical path, you can try synthesizing only the BoomTile

glzhou97, 2024-06-08:
I noticed that there are several clocks for Boom, namely [clock, auto_prci_ctrl_domain_tileResetSetter_clock_in_member_allClocks_subsystem_cbus_0_clock, auto_prci_ctrl_domain_tileResetSetter_clock_in_member_allClocks_subsystem_mbus_0_clock,...
, debug_clock]. May I ask what are the proper timing constraints for each clock? If I want to compute CoreMark/Mhz, which critical path slack shall I use to calculate that? Shall I use clock? Or the WNS amount all the clock?

glzhou97, 2024-06-08:
If anyone can share their timing constraints file would be highly appreciated!