riscv-boom/riscv-boom #477: O3Pipeview trace for BOOMv3

owahidcse99, 2020-07-20:
Feature request

**Impact**: no rtl change | rtl refactoring | new rtl | unknown
New rtl

**Development Phase**: request | proposal
Request
**Other information**

For the +verbose option, BOOMv2 used to print out pipetrace if
val O3PIPEVIEW_PRINTF   = true            in boom/src/main/scala/consts.scala.
I had very old post regarding that
https://groups.google.com/forum/#!searchin/riscv-boom/pipetrace%7Csort:date/riscv-boom/tEhKePUtVCw/NV6aq3TjCgAJ

Looking at chipyard/generators/boom/src/main/scala/common/consts.scala
there is no option for that. Neither "recursive grep" in Chipyard shows up anything. Does it mean, BOOMv3 would not print out pipetrace? This is in fact very useful to look what is happening inside pipeline stages.

However in BOOMv3 (similar to BOOMv2) the pipeview helper exists that used to convert raw traces to  gem5 compatible
chipyard/generators/boom/util/pipeview-helper.py.
Therefore, to convert raw pipetrace, pipeview helper exists, but no way of generating raw pipetrace at all in BOOMv3.

Also, is it the case that Rocket core can't generate pipeline trace as well? I haven't find anything in below as well.
chipyard/generators/rocket-chip/
Thanks!

**What is the current behavior?**
No pipeview trace in BOOMv3
**What is the expected behavior?**
Pipeview trace
**Please tell us about your environment:**

**What is the use case for changing the behavior?**

jerryz123, 2020-07-28:
The pipeline trace feature was temporarily removed when the pipeline was restructured. I will add it back

guyue555, 2021-04-06:
is this feature added? i have not find it also.