riscv-boom/riscv-boom #478: How to enable RSB in the latest boom

hz1490919302, 2020-07-24:
**Type of issue**: bug report | feature request | question | other enhancement

**Impact**: no rtl change | rtl refactoring | new rtl | unknown

**Development Phase**: request | proposal

**Other information**

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

jerryz123, 2020-07-24:
What do you mean,? RAS works in default configuration

hz1490919302, 2020-07-29:
Comments I saw elsewhere：
https://github.com/riscv-boom/boom-attacks/issues/3#issuecomment-561804329

jerryz123, 2020-07-29:
That issue has since been resolved.

hz1490919302, 2020-07-29:
I was able to successfully run the conditional branch mispredict and the indirect branch mispredict attacks with this chipyard  and the SmallBoomConfig. However the return stack buffer attack did not recover the correct secret. Therefore, before you reply, I suspect that RAS is not working.

jerryz123, 2020-07-29:
I am certain the RAS works. The problem is that we did not finish the implementation of the RAS attack in that re

closed 2020-07-29 by jerryz123