riscv-boom/riscv-boom #482: Any information about SFB?

hyf6661669, 2020-08-05:
I find there are some codes about ```SFB```, which seems have something to do with a small loop that fall in a whole Icache line. But the document doesn't tell any details about it. Could you please share any information?

jerryz123, 2020-08-05:
See section 4.2 here, https://carrv.github.io/2020/papers/CARRV2020_paper_15_Zhao.pdf

SFB is an acronym for short-forwards-branch, which is a subset of branches for which a special optimization is applied, in which the hardware unconditionally executes both directions of the branch, potentially avoiding an expensive mispredict.

The frontend marks branches over a short sequence (<32bytes) of non-excepting integer instructions as "sfb" branches. A "sfb" branch, instead of resolving a direction, sets a physically renamed predicate register with the direction of the branch. Instructions in the shadow of the "sfb" branch read the predicate register to determine if they should architecturally execute (the branch is not taken), or if they should not architecturally execute (the branch is taken). 

In the latter case, due to register renaming, the shadow instruction must still perform a copy between physical registers.

sorear, 2020-08-06:
There seem to be two ways to implement something like this:

I would be interested to read anything that exists on the SFB design space exploration.

What got implemented in BOOM explictly renames predicates, performs predication in the ALU execution stage, and always reads the same registers.  Since the ALU is limited to reading 2 registers and the old destination register must be read, predication is only supported on simple ALU instructions that are either destructive (rs1=rd) or use an immediate operand.  Since the registers that are read do not depend on the predicate value, the predicated instruction can issue immediately after the predicate-generating instruction.  Since loads and stores cannot be modified and could trap if performed unconditionally, a large fraction of short branches cannot be converted even with compiler assistance.

How I expected this to work before reading it: Predicates are implicitly renamed (explicit renaming is an important optimization to avoid multiple 64-bit result buses, but for a 1-bit predicate this is much less of an issue.)  Since the predicate value is available in the issue stage, it can be used to select which registers to read and what execution unit to issue to, so any instruction can be subject to predication (probably except branches; all integer, float, memory, and system instructions should be fine though). There's a 1 or 2 cycle penalty on the latency _if the predicate is the limiting input_ but no dependency on the false path for values.

jerryz123, 2020-08-06:
> What got implemented in BOOM explictly renames predicate

Explicit renaming of predicates was very cheap to implement. Since the implementation guarantees only 1 SFB can be fetched per cycle, the renamed physical register index reuses the fetch-target-buffer index for the short branch.

> Since loads and stores cannot be modified and could trap if performed unconditionally, a large fraction of short branches cannot be converted even with compiler assistance.

I ignored loads and stores in the initial implementation both for simplicity, and because the GCC RISC-V target does not support optimizing for short-branches over loads/stores.

I don't believe extending the current implementation to support loads/stores is especially complex. Address generation to the LSU can be delayed until the predicate is known, at which point any generated exception can be masked off, and the LDQ/STQ entries can be marked as empty.

> Since the predicate value is available in the issue stage, it can be used to select which registers to read and what execution unit to issue to, so any instruction can be subject to predication 

Supporting predication for all instructions still imposes a large cost on the issue-queues, as a 2-operand instruction in a issue queue would require three wakeup ports. 2 for the operands, and 1 for the stale physical register.

sorear, 2020-08-06:
> Supporting predication for all instructions still imposes a large cost on the issue-queues, as a 2-operand instruction in a issue queue would require three wakeup ports. 2 for the operands, and 1 for the stale physical register.

I may be a little bit confused about what part of the design "wakeup port" refers to but core.scala computes the number of wakeup ports passed to the ROB constructor as the number of execution ports requiring integer writeback, with no input from the issue slots.  You would need 3 comparators on each wakeup port rather than the current 2, but that's entirely local logic within the issue slot.

The original BOOM SFB proposal I made in May 2017 avoided having a pred_wakeup_port entirely by passing predicate data over the brupdate bus, but this is probably less compelling since branch masks kind of want to be unary.

jerryz123, 2020-08-07:
> You would need 3 comparators on each wakeup port rather than the current 2, but that's entirely local logic within the issue slot.

This is what I mean. Sorry, I think your terminology is more correct than mine. It seems excessive to add an additional comparator for every single issue slot just to support slight additional flexibility for short-forward-branches.

> The original BOOM SFB proposal I made in May 2017 avoided having a pred_wakeup_port entirely by passing predicate data over the brupdate bus, but this is probably less compelling since branch masks kind of want to be unary.

Did you post this proposal to the mailing list? I can't seem to find it.

sorear, 2020-08-07:
That would have been the reasonable thing to do, but no.  I sent it to Chris on IRC and then completely forgot about it until reading about the SFB feature recently.  I take full responsibility for losing this, but it also wasn't a very fully done proposal.

```
﻿[2017-05-11 21:21:47] <sorear> the user spec talks about designs where the frontend detects low-confidence branches with short shadows and if-converts them on the fly.  does anything like that make sense for real applications in 2017?
[2017-05-11 21:22:51] <boomcpu> i suspect a lot of algorithms have a pattern of If (cond) a = b
[2017-05-11 21:23:02] <boomcpu> i believe graph algorithms have stuff like that
[2017-05-11 21:23:19] <boomcpu> "travel to every node, do a small amount of computation and the branches are all data-dependent"
[2017-05-11 21:23:33] <sorear> yeah, including stuff like if (cond) *a = b which can't be handled by conditional moves on x86
[2017-05-11 21:23:55] <boomcpu> i don't have a good feel for what the complexity to do it is, but it's an interesting thought
[2017-05-11 21:24:26] <sorear> arm7 of course has safe conditional stores
[2017-05-11 21:24:47] <boomcpu> yah
[2017-05-11 21:25:30] <boomcpu> interestingly, it made the 462.libquantum data for arm7 look terrible compared to x86 and risc-v, b/c the compiler used the conditional store, even though the branch was hihgly predictable, so it was extra work.
[2017-05-11 21:26:21] <boomcpu> doesn't necessarily translate to more or less cycles, but interesting nonetheless
[2017-05-11 21:27:02] <sorear> commentary at the end of §2.5
[2017-05-11 21:27:46] <sorear> yeah it's one of those cases where making the wrong decision statically is easy
[2017-05-11 21:29:44] <boomcpu> ah, i didn't notice they provided citations. neat.
[2017-05-11 21:29:47] <sorear> related question, how much extra work is it to modify a branch predictor to output "taken / not taken / this branch seems pretty much random so it's worth doing other things with"
[2017-05-11 21:32:45] <boomcpu> interesting question. i don't know how to quantify "extra work", but it's certainly something that could be tracked to some degree. very advanced predictors are likely to have some sort of "confidence" or "usefulness" bits associated with prediction entries

[2017-05-12 05:07:54] <sorear> impact of auto-predication on a boom-style backend could be rather small. I don't think you need any new register read ports or wakeup ports
[2017-05-12 05:08:41] <sorear> impact on the frontend is harder for me to estimate at my current depth of understanding…
[2017-05-12 05:09:37] <sorear> i found a substantially shorter implementation for round/trunc/floor/ceil/rint/nearbyint/llrint on RV32G, what should I do with it
[2017-05-12 05:10:14] <sorear> (for all except llrint it's shorter than the RV64G impl too, but by a smaller factor)
[2017-05-12 05:10:47] <sorear> guess I could ticket against riscv-glibc
[2017-05-12 05:11:22] → proteus-guy joined (~proteus-g@180.183.250.20)
[2017-05-12 05:18:52] <sorear> -sketch follows, none of this is to be taken as a request for work-
[2017-05-12 05:21:53] <sorear> → spend more time looking at the frontend before posting any weird ideas
[2017-05-12 05:40:37] <sorear> but the basic idea is that issue slots can have a "predication branch mask" as well as the "speculation branch mask (br_mask)", and then use the existing kill and branch resolution networks (no new long wires) to deliver predication branch masks
[2017-05-12 05:40:53] <sorear> an instruction cannot issue as long as it has a bit set in pred_mask
[2017-05-12 05:42:26] <sorear> when the kill network delivers a bit that matches pred_mask, the instruction is not *entirely* killed but rewrites into a mv or fmv from stale_pdst to pdst
[2017-05-12 05:43:00] <sorear> and then issues (when stale_pdst is ready), to cancel out the effect of the rename
[2017-05-12 05:47:02] <sorear> > // ALU Unit
[2017-05-12 06:08:58] <sorear> … this is shockingly simple.  one more thing for the "things I'll try to do if I get a fpga setup" pile
[2017-05-12 06:17:21] <boomcpu> sorear an interesting idea, i'll have to ponder it a bit more.
[2017-05-12 06:22:17] <sorear> predictability prediction can be done with a sidecar predictor that's fed with "did TAGE get this right" instead of "was the branch taken"; it could probably get good results even much smaller.  (Or it could be rolled into the main predictor)
[2017-05-12 06:22:52] <sorear> the "branch unit" needs to know that converted branches don't redirect PC ever (although they still need to send resolve/kill signals)
[2017-05-12 06:23:18] <sorear> commit can treat converted and unconverted branches precisely the same, they both update the predictor
[2017-05-12 06:23:38] <sorear> the only part where I see potential dragons is decode
[2017-05-12 06:24:39] <sorear> in particular, converting a branch shadow which contains conditional branches of its own is likely to not work, so the decode "is this a candidate" logic likely needs to look ahead a few instructions(??)
[2017-05-12 06:24:44] <sorear>  / if any of that is helpful
[2017-05-12 20:55:55] <sorear> boomcpu: did you get everything up to "/ if any of that is helpful" prior to disconnecting?
[2017-05-12 20:58:43] <boomcpu> sorear the last message i got was 3:23, "the only part ... dragons is decode"
[... cut ...]
[2017-05-12 21:48:50] <sorear> applying a fudge factor and saying 50% because risc-v doesn't even have cmov, so a larger fraction of convertable branches would exist
[2017-05-12 21:50:08] <sorear> there's some overlap here with the fusion stuff.  what is c.bnez a0,pc+2; c.mv a1,a2 but a 4-byte cmov
[2017-05-12 21:50:58] <boomcpu_> interesting
[2017-05-12 21:54:48] <sorear> predicated commit would probably have better ipc than predicated issue, but it'd also be substantially more complicated
```

jerryz123, 2020-08-07:
Your proposal is pretty spot-on accurate to what was implemented. Thanks for digging this up.

> predictability prediction can be done with a sidecar predictor that's fed with "did TAGE get this right" instead of "was the branch taken"; it could probably get good results even much smaller.  (Or it could be rolled into the main predictor)

This is a good point, ideally you'd only want to apply predication to regularly mispredicted branches. In the current implementation, predicted-not-taken branches optimized into SFBs do not update the branch predictors when they resolve. The consequence of this is that a unpredictable SFB would naturally cause the predictor to settle on strong-not-taken, such that future fetches to this branch would go down the not-taken, optimizable path.

> so the decode "is this a candidate" logic likely needs to look ahead a few instructions

Yeah, this was the reason why the limitation is 32-bytes of shadowed instructions, (2 fetches). A potential branch in fetch-4 checks younger instructions in fetch-4 and fetch-3 to see if all instructions in the branch shadow are candidates for predication. 

I guess more accurately, its actually max 30-bytes (32 - 2byte compressed branch) assuming the branch is aligned to the start of the first fetch packet.

I'm curious what the potential performance improvement is with supporting predicated loads/stores. Last time I checked, there weren't too many critical sections where this would help a lot, but its likely some compiler support is needed to fully capitalize on this feature.

closed 2021-05-01 by jerryz123