riscv-boom/riscv-boom #483: Question about boom

JaewonHur, 2020-08-12:
Hi
Recently I'm studying the `frontend` implementation of boom.
Especially I'm looking about prediction and update logic.

And, I found `BranchChecker` does update `RAS` when mispredicted `call` instruction is detected,
but it does not update `RAS` when `ret` instruction.
(Update for `ret` instruction was only in right after `BTB` predicts `return`.)

Is there any reason why `BranchChecker` does not update `RAS` when `ret` instruction?

jerryz123, 2020-08-12:
The `BranchChecker` is from the old implementation of the frontend. Please look at the current predictor implementation in BOOMv3 for an updated design.

JaewonHur, 2020-08-12:
Ah, thanks.
By the way, is there any protection mechanism related to Spectre or Meltdown in BOOMv3?

jerryz123, 2020-08-12:
No. Proposed schemes generally have non-negligible performance or complexity cost, so they were not implemented by default in SonicBOOM.

closed 2021-05-01 by jerryz123