riscv-boom/riscv-boom #487: Q

Lingrui98, 2020-08-27:
**Type of issue**: question

I'm currently trying to test BOOM for its branch prediction performance. In order to get the output, I tried to enable the o3-pipeview trace but found it removed in #477 . Then I decided to use the hardware performance counters based on HPE, and found that the HPEs related to branch prediction was commented after this commit 
> 2cd16ed23f4d6af696be6cf517779b4ec2430a17

```    new freechips.rocketchip.rocket.EventSet((mask, hits) => (mask & hits).orR, Seq(
//      ("I$ blocked",                        () => icache_blocked),
      ("nop",                               () => false.B),
      // ("branch misprediction",              () => br_unit.brinfo.mispredict),
      // ("control-flow target misprediction", () => br_unit.brinfo.mispredict &&
      //                                             br_unit.brinfo.cfi_type === CFI_JALR),
      ("flush",                             () => rob.io.flush.valid)
      //("branch resolved",                   () => br_unit.brinfo.valid)
    )),
```
It seems that this commented code does not fix in current version of BOOM. I wanted to re-enable this feature and changed it to this:
```
    new freechips.rocketchip.rocket.EventSet((mask, hits) => (mask & hits).orR, Seq(
      ("I$ blocked",                        () => icache_blocked),
      ("nop",                               () => false.B),
      ("branch misprediction",              () => brupdate.b2.mispredict),
      ("control-flow target misprediction", () => brupdate.b2.mispredict &&
                                                  brupdate.b2.cfi_type === CFI_JALR),
      ("flush",                             () => rob.io.flush.valid),
      ("branch resolved",                   () => b1.resolve_mask.orR)
    )),
```
Am I getting the conditions right in correspond with their names?
In addition, does BOOM send only one CFI for BPD to update in a cycle, or the BPD can update multiple CFIs in a cycle?

jerryz123, 2020-08-27:
Since BOOM can retire multiple CFIs per cycle, that interface for HPEs would yield inaccurate results. Additionally, it would be necessary to count only the prediction result of committed branches.

Generally, what I do is enable a committed-branch print log with this config fragment: https://github.com/riscv-boom/riscv-boom/blob/f69eac89a2ae849cec0b157f3b6812f447408656/src/main/scala/common/config-mixins.scala#L36-L43
Then, the emitted log can be post-processed with the script in boom/util/branch-processor.py to generate statistics of prediction accuracy on frequently seen branches.

ccelio, 2020-08-27:
For a future resolution, I would recommend counting branch mispredicts during Execute by looking at what comes out of the BranchResolutionUnit. It will over-count mispredicts since it includes wrong-path mispredicts, but that's okay enough for getting this visibility in hardware (and thus software), and it's not too far off from the retirement count.

Or another strategy (not mutually exclusive) is to count front-end resteers/restarts, and bucket the restarts into different blame categories (e.g., restart-from-br-exe-unit-that-tagetable3-had-predicted), and then track "mispredicts per 1k instructions".

Lingrui98, 2020-08-28:
Thanks for replying! Which branch on chipyard should I use? I simply checkout to dev from master but got compile errors like this:
```
[error] /home/glr/boom/chipyard/generators/boom/src/main/scala/common/config-mixins.scala:27:8: not found: value TilesLocated
[error]   case TilesLocated(InSubsystem) => up(TilesLocated(InSubsystem), site) map {
[error] 
```

jerryz123, 2020-08-28:
If you switch chipyard versions, you must also update all of the submodules with the init-submodules script, or else there will be version incompatibilities.

Lingrui98, 2020-08-28:
I updated all the submodules on branch #dev, and followed the instructions in the document as below:
```bash
make CONFIG=MegaBoomConfig run-binary-debug BINARY=mytest.riscv

# The below command does the same thing
./simulator-chipyard-MegaBoomConfig-debug +verbose mytest.riscv
```
But these commands does not seem to work now
```log
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not open +verbose (did you misspell it? If VCS, did you forget +permissive/+permissive-off?)
```
What commands should I use to enable all chisel prints?

I modified BoomConfigs.scala as below:
```scala
class MegaBoomConfig extends Config(
  new boom.common.WithNMegaBooms(1) ++                           // mega boom config
  new boom.common.WithBoomBranchPrintf ++
  new boom.common.WithBoomCommitLogPrintf ++
  new chipyard.config.AbstractConfig)
```
Am I getting things right? Should I add the 
```scala
  new boom.common.WithBoomBranchPrintf ++
  new boom.common.WithBoomCommitLogPrintf ++
```
in config-mixins.scala or anywhere else?

jerryz123, 2020-08-28:
You need to add `WithBoomPrintf` lines *above* the `WithNMegaBooms` line

Lingrui98, 2020-09-02:
I tested coremark(riscv64-baremetal) with 1 iteration and get the following result:
[image]
The mispredict rate is about 6%, but the closed issue #119 says BOOM with TAGE should get >96% on Coremark. It seems this result conflict with this assertion.
(I ran the test with default MegaBOOMConfig)

jerryz123, 2020-09-02:
Naturally the misprediction rate will be higher after only 1 iteration of coremark.

closed 2022-02-24 by Lingrui98