riscv-boom/riscv-boom #491: lrsc bug in boom?

JaewonHur, 2020-09-14:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: proposal

I tried to use boom tag v3.0.0 in chipyard, but I was not able to build the simulation due to some 
unresolved dependency. 
So I'm now using boom  commit 3ea630a4.

Anyway there is a confusing behavior related to lrsc instructions.

As in the test file in bug.zip,
when lr.w inst. first reserve some address (say d3.) and following sc.d try to save data in address (d3 + 8),
Boom succeeds but spike fails.

I think Boom should fail since address (d3 + 8) is not in the reservation set.

[bug.zip](https://github.com/riscv-boom/riscv-boom/files/5215858/bug.zip)

jerryz123, 2020-09-14:
The ISA manual states

> LR loads a word from the address in rs1, places the sign-extended value in rd, and registers a reservation on the memory address. SC writes a word in rs2 to the address in rs1, provided a valid reservation still exists on that address.

> An implementation can reserve an arbitrary subset of the memory space on each LR

Both BOOM and Rocket reserve the cache line containing the address of the LR. This means a subsequent SC to a different address in same cache line will succeed, but this does not violate the spec. 

You should be able to demonstrate the same behavior in Rocket.

JaewonHur, 2020-09-14:
Thanks, I misunderstood the second quote.

closed 2020-09-14 by JaewonHur