riscv-boom/riscv-boom #495: A non-technical question: Thank you for taking the time to answer it

hz1490919302, 2020-10-15:
I'm a new student studying Chisel and I'm looking at the Boom source code. Because Boom has so many very advanced implementations, I can't fully understand the meaning of some of the concrete code, such as how Config.Scala works. 
I really admire the BOOM author's coding prowess.
I have read the BOOM development document. May I ask how Can I quickly understand the BOOM implementation details? Thank you very much

jerryz123, 2020-10-15:
One way to do this is by trying to identify the registers and pipeline stages an instruction passes through as it executes. You can execute a simple hello-world program in RTL simulation, generate a waveform, and manually trace the execution of the program through the waves. It is tedious initially, but you will quickly learn the correspondence between the microarchitecture and the source RTL.

Some of the source code in this repo (Config.scala, for example) contains code to aid in SoC generation and integration, rather than concrete RTL. To understand how these systems work, the Chipyard documentation is a good place to start. https://chipyard.readthedocs.io/en/latest/

hz1490919302, 2020-10-15:
Thank you very much for your reply. I will run a HelloWorld program to learn Boom. You're a great developer. Thank you again.

hz1490919302, 2020-10-30:
Hello, I have now built the **chipyard** project and generated a BOOM simulator **simulator-chipyard-SmallBoomConfig**. Then I wrote the **hello.c** file with only one printf statement and compiled it to form the **hello.riscv** executable file. Finally, I used the command `make CONFIG=SamllBoomConfig run-binary-debug BINARY=hello.riscv `to generate a **hello.chipyard.TestHarness.SmallBoomConfig.vcd** file.

I use the `gtkwave hello.chipyard.TestHarness.SmallBoomConfig.vcd` command to view the waveform. But I don't understand the meaning of the various wires or registers in the figure below, and I can't connect it with hello.c. What should I do?
x-special/nautilus-clipboard
copy
file:///home/hezhuo/Pictures/2020-10-30%2016-36-10%E5%B1%8F%E5%B9%95%E6%88%AA%E5%9B%BE.png

hz1490919302, 2020-10-30:
Hello, I have now built the chipyard project and generated a BOOM simulator **simulator-chipyard-SmallBoomConfig**. Then I wrote the **hello.c** file with only one printf statement and compiled it to form the **hello.riscv** executable file. Finally, I used the command `make CONFIG=SamllBoomConfig run-binary-debug BINARY=hello.riscv` to generate a **hello.chipyard.TestHarness.SmallBoomConfig.vcd** file.

I use the `gtkwave hello.chipyard.TestHarness.SmallBoomConfig.vcd `command to view the waveform. But I don't understand the meaning of the various wires or registers in the figure below, and I can't associate it with hello.c. What should I do?
Looking forward to your answer, thanks.

[image]

jerryz123, 2020-10-30:
The SST panel displays the module hierarchy of the simulated system. You should navigate that hierarchy until you find the BoomTile. The rough hierarchy is TestHarness->ChipTop->DigitalTop->BoomTile. Within BOOM Tile there are separate modules for instruction fetch (frontend), backend, and data-cache.

I recommend starting by looking at the rob.io.commit signals in the core. These should show the sequence of instructions being committed by the core, which you can match to your program.
I would also adding the WithBoomCommitLogPrintf option to your run so you can get a debug trace of all instructions committed.

hz1490919302, 2020-11-03:
According to your guidance, I have found the BoomTile. And I have added the WithBoomCommitLogPrintf option to my run. I got the following **hello.out** file. However, I don't understand the meaning of these printouts, and I haven't found relevant descriptions in the documentation. Can you explain their meaning? Thank you very much.
[image]

jerryz123, 2020-11-03:
This is the trace of committed instructions.
The format is privilege level, PC, insn, rd, wdata.

closed 2020-12-09 by hz1490919302