riscv-boom/riscv-boom #498: Does boom-v3 implement any mitigation of spectre?

dodo98765x, 2020-11-06:
I have surveyed a paper about reproduced spectre on Boom in https://carrv.github.io/2019/papers/carrv2019_paper_5.pdf. After I run the spectre code(condBranchMispred.c) provided by this paper on boom v3 and boom v2.2.3 (config = LargeBoom).The result shows the spectre code can successfully attack on Boom v2.2.3 but failed on Boom v3 except for the first byte. So I am wondering if there are any mitigation(ex. L0 Buffer) in Boom v3.

Thank you so much for your reading and your kind help!

dodo98765x, 2020-11-20:
I have edited the spectre code with one byte secret data and run it on Boom-v3. The result shows that the secret data will be attacked. Attached is my spectre C code.
[condBranchMispred_1Byte.zip](https://github.com/riscv-boom/riscv-boom/files/5571145/condBranchMispred_1Byte.zip)

jerryz123, 2020-11-20:
Thanks for trying this out. You are correct there are no Spectre mitigations in boomv3.

dodo98765x, 2020-11-20:
Thanks for your reply.

I found that if I edit the spectre code with three Bytes secret data, the second and third Byte of the secret data won't be attacked on Boom v3. So I'm wondering if there are any difference between Boom 2.2.3 and Boom v3 that caused it ? Attached is my spectre C code.
[condBranchMispred_3Byte.zip](https://github.com/riscv-boom/riscv-boom/files/5571442/condBranchMispred_3Byte.zip)

jerryz123, 2020-11-20:
Boomv3 has different L1 DCache behavior. Notably, speculatively issued cache misses that are squashed quickly enough may abort the refill. This may affect the behavior of some Spectre sample attacks, but it is not a mitigation.

dodo98765x, 2020-11-20:
Is it means that Boom v3 has less penalty when L1 DCache miss? If so, it seems like spectre will easier to speculatively load data to L1 DCache. I am wondering why will it abort the refill?

jerryz123, 2020-11-20:
The penalty for a cache miss is the same. 

The difference is that in v3 a speculatively issued load that is immediately squashed can abort the refill process before completion, avoiding an unnecessary cache line eviction.

dodo98765x, 2020-11-20:
Is it mean that there is a buffer before L1 Dcache to temporary storage the refill data and the data will be squashed if CPU found the speculatively issued load should not be executed?

jerryz123, 2020-11-20:
Yes.

dodo98765x, 2020-11-22:
It is sounds like the "L0 speculation buffer" in this paper https://carrv.github.io/2019/papers/carrv2019_paper_5.pdf
Am I right?

dodo98765x, 2020-11-22:
Thank you so much for your clear and unambiguous explanations on this topic. They have helped me tremendously!

lihao1915, 2020-12-04:
> Thank you so much for your clear and unambiguous explanations on this topic. They have helped me tremendously!

Excuse me, do you know how to reproduce a whole Spectre V1 attack on boom V3? Thank you very much!!

dodo98765x, 2020-12-04:
> > Thank you so much for your clear and unambiguous explanations on this topic. They have helped me tremendously!
> 
> Excuse me, do you know how to reproduce a whole Spectre V1 attack on boom V3? Thank you very much!!

Hi,

To reproduce spectre on boom v3, first you should compile the boom to generate a simulator by Chipyard. Then compile the spectre code to ELF. Finally use the simulator to execute the ELF. Attached are Chipyard and the spectre code I have found on github.

Chipyard : https://github.com/ucb-bar/chipyard
boom-attacks :  https://github.com/riscv-boom/boom-attacks

lihao1915, 2020-12-06:
> Hi,
>
> To reproduce spectre on boom v3, first you should compile the boom to generate a simulator by Chipyard. Then compile the spectre code to ELF. Finally use the simulator to execute the ELF. [...]

Hi,
Thank you very much!
I met the same problem with u: only the first byte of a string could be attacked successfully. So if I want to attack the whole string( may be 16 bytes), what should I do ?

lihao1915, 2020-12-08:
> The penalty for a cache miss is the same.
> 
> The difference is that in v3 a speculatively issued load that is immediately squashed can abort the refill process before completion, avoiding an unnecessary cache line eviction.

Excuse me, if the cache performance like you said above, the first byte should also not be attacked successfully. However, the first byte is attacked successfully, so why the remainder can not be attacked?