riscv-boom/riscv-boom #500: How can I use boom to build a multicore system?

leizhangpku, 2020-11-13:
**Type of issue**: bug report | feature request | question | other enhancement

**Impact**: no rtl change | rtl refactoring | new rtl | unknown

**Development Phase**: request | proposal

**Other information**

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

jerryz123, 2020-11-13:
Please see documentation in the Chipyard repository for instructions on building multicore boom systems

leizhangpku, 2020-11-13:
> Please see documentation in the Chipyard repository for instructions on building multicore boom systems

Ok, thank u.

> Please see documentation in the Chipyard repository for instructions on building multicore boom systems