riscv-boom/riscv-boom #503: fcsr 'invalid operation' flag is not set

JaewonHur, 2020-11-24:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: proposal

**Other information**

Hi, there is a mismatch between spike (rocket) and boom when doing `fdiv` operation.
In the test case below, spike and rocket set `Invalid Operation` flag in `fcsr`, 
but boom does not set.

Here is test case.
The buggy assembly file is `test_p.S` and you can make binary (`test_p.riscv`) after running `make` 
Boom v2, v3 all affected.

[test.zip](https://github.com/riscv-boom/riscv-boom/files/5590334/test.zip)

jerryz123, 2020-12-01:
Thanks, there were some bugs with the implementation of BOOM's FDivSqrt unit. You can check out the fixes in the `mem2` branch.

JaewonHur, 2020-12-10:
Thanks,

close the issue.

closed 2020-12-10 by JaewonHur

zhangkanqi, 2023-11-02:
But this bug seems to be still unfixed.