riscv-boom/riscv-boom #504: Misaligned load reserve should not set reservation

JaewonHur, 2020-11-30:
**Type of issue**: bug report 

**Impact**: rtl refactoring

**Development Phase**: proposal

**Other information**

Misaligned load reserve instruction should not set reservation.
However boom set reservation even after misaligned load exception.

`make test_p.riscv` makes the bug case binary (`test_p.riscv`).

Logs of both boom and rocket are also included.

[test.zip](https://github.com/riscv-boom/riscv-boom/files/5615656/test.zip)

JaewonHur, 2020-11-30:
It seems strange,

When just misaligned `lr.d` and correctly aligned `sc.w` are executed,
boom the store conditional does not succeed (as expected.)

But if `amo` operation accesses the same cache line at first, 
`sc.w` succeeds (bug).

[testcases.zip](https://github.com/riscv-boom/riscv-boom/files/5615710/testcases.zip)

jerryz123, 2020-11-30:
Thanks for the detailed report.
Can you post a waveform from BOOM demonstrating the issue? I am having trouble reproducing the behavior.

JaewonHur, 2020-12-01:
Attach all assembly, binary and waverforms.

[tests.zip](https://github.com/riscv-boom/riscv-boom/files/5619616/tests.zip)

jerryz123, 2020-12-01:
The target address `0x8001010c` is word-aligned, which causes `amoand.w` to succeed, but `lr.d` to fail. BOOM does not support hardware misaligned loads/stores, including atomics and LR/SC.

jerryz123, 2020-12-01:
I'm not sure what the error is here? In bug.vcd, the core appears to take an exception on a misaligned `lr.d`, which is expected.

JaewonHur, 2020-12-01:
yes
'lr.d' fails as expected but the following 'sc.w' succeeds.

Following 'sc.w' should fail since 'lr.d' failed.

jerryz123, 2020-12-01:
Oh I see, you are right, this is a bug. Sorry for the confusion.

Loads which translate successfully, but still generate exceptions, were still being fired into the L1 DCache. For normal loads this would not be an issue, as loads don't leave behind architecturally visible side effects. For LR, however, we need to specially avoid acquiring a reservation in this event.

JaewonHur, 2020-12-01:
Thanks.
But the bug is only triggered when `amo` operation accesses the same cash line 
before `lr/sc` instructions.
It should be checked. 

By the way, could you confirm my other bug reports? 
It seems `hardfloat` contains some bugs.

jerryz123, 2020-12-01:
Any operation which caches the line of the LR will cause the misaligned LR to set a reservation, it doesn't have to be AMO. 

Thanks, I will look at the floating point issues soon.

JaewonHur, 2020-12-01:
Ah, thanks.

closed 2021-05-01 by jerryz123