riscv-boom/riscv-boom #506: Programming question

hz1490919302, 2020-12-09:
Hello. Is there any method in Boom that can make an instruction stop for several cycles in the execution phase when it encounters condition _B_, and then continue to execute, that is

```
if(B)

stall-a-instruction-for-some-time
```

I couldn't find a way to stop the instruction

jerryz123, 2020-12-09:
So stalling an instruction in the execution (exu) stage is a tricky. The (iss//rrd/exu) stages are pipelined, so stalling an instruction in exu means also stalling instructions behind it in iss/rrd. Additionally, since wakeup signals are generated for bypassable instructions in iss, an instruction in exu may have dependent instructions in other parts of the core waiting for the instruction to wb in the next cycle. Given the complexity with dealing with these cases, there is no generalized mechanism for arbitrarily stalling an instruction in exu currently.

However, for some specific instructions, this "stall in execute" behavior is necessary. For example, fdiv instructions occupy the iterative fdiv functional unit for an uncertain amount of cycles, so they appear to "stall" in execute until the fdiv unit completes. These cases are handled by extra scheduling logic which enforce that there aren't other issued instructions which have data or structural dependencies on an instruction "stalled" in fdiv.

The only way currently to "stall" an arbitrary instruction is to stall it in dispatch. You can see the dec_stalls vector in core.scala which checks a bunch of conditions for stalling an instruction here.

hz1490919302, 2020-12-10:
Thank you very much for your answer. It's very useful.

closed 2021-09-29 by jerryz123