riscv-boom/riscv-boom #510: Question about Boom memory operation tests

JaewonHur, 2020-12-22:
**Type of issue**: question

I wonder how the memory operation of Boom in multi-core environment is tested?
Are there a sort of unit tests like `riscv-test` for multi-core testing?
or are there any other methods to test boom memory coherency and consistency model?

jerryz123, 2020-12-22:
riscv-tests can be modified to run across multiple cores, although the coverage of memory-coherency/consistency corner cases is quite poor.
The most rigorous testing we do on the memory system is through the `tracegen` utility in Chipyard/Rocketchip. `tracegen` generates random sequence of load/store/amo for each client cache in a multi-core system, and can be configured to stress-test memory-consistency corner cases.

JaewonHur, 2020-12-22:
Thanks.
Then how's the bug is detected, since it is a sort of semantic bug ?
Do we have to manually analyze the cases ?

jerryz123, 2020-12-22:
The traces emitted by `tracegen` can be analyzed by Axe: https://github.com/CTSRD-CHERI/axe to check for validity according to a memory consistency model. We target WMO. 

I believe you can do `make tracegen CONFIG=BoomTracegenConfig` in Chipyard to do a tracegen simulation, and then have Axe check the resulting traces.

JaewonHur, 2020-12-22:
Thanks a lot.
It seems very useful.

closed 2021-05-01 by jerryz123