riscv-boom/riscv-boom #524: Memory coherency bug in Boom

JaewonHur, 2021-01-24:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: proposal

**Other information**

Hi, 
I'm testing Boom (& Rocket chip) memory system on chipyard v1.3,
and it seems there's a bug.

The minimized AXE trace is as follow,
```
1: M[6] := 497 @ 8699:
0: M[5] := 426 @ 8820:
0: sync @ 8821:8864
0: M[6] == 497 @ 8866:8965
1: M[6] := 505 @ 8890:
1: sync @ 8891:8892
1: M[5] := 511 @ 8896:
1: { M[5] == 426; M[5] := 525} @ 9124:
```

Where, write on `M[5]` at 7th line is ignored when reading `M[5]` at 8th line.

Would it be useful to give you the minimized input trace (for debugging) ? 
or just the AXE trace is enough?

(Actually I made some modification on the `tracegen` so it would take some time to give the bug reproduce)

jerryz123, 2021-01-24:
Thanks for finding this. Are you able to provide a waveform? That would make it easiest to find the issue

JaewonHur, 2021-01-26:
Here's the full axe trace and waveform.
Thanks.

[boom_bug.zip](https://github.com/riscv-boom/riscv-boom/files/5870654/boom_bug.zip)

jerryz123, 2021-01-26:
I may be wrong, but I believe this is actually a bug with the implementation of the TraceGen trace generator in RocketChip. The key to this are the `sync` lines. These should perform the equivalent of the RISC-V `fence` instruction.

I believe the bug is that the implementation of issuing a `sync` or `fence` operation in the TraceGen testbench does not actually perform a fence, it merely waits for all prior loads and stores to be completed. Notably, it does not wait for inflight stores in a store buffer to drain.

When the actual Rocket or BOOM cores execute the `fence` instruction, they stall until the dcache or load-store-unit are empty, producing correct, consistent behavior. For tracegen to behave correctly, there probably needs to be a signal exposed from the cache/load-store-unit into the TraceGen unit, stalling a fence until the cache/lsu is ordered.

Thanks for finding this very obscure bug.

jerryz123, 2021-01-26:
FYI, if you are trying to get tracegen to generate adversarial traces, I recommend generating traces with only Loads and Stores, and not any fences, LR, SC, or swap operations. The LSU takes a pretty blunt approach to these. Since these instructions are relatively infrequent in code compared to normal loads/stores, its safe, and not too costly to just stall for these instructions.

Generating traces with only normal loads and stores maximizes OOO execution, and stresses the most corners cases for the logic which handles cache management/memory consistency/memory coherency.

JaewonHur, 2021-01-26:
Thanks for check and comments. 
I see that instructions other than load and store are not so important.

I have one more question. Are there any measure you or your team (?) uses to check the memory system is sufficiently tested on dynamic verification?

jerryz123, 2021-01-27:
We test booting Linux with multicore configs.

closed 2021-05-01 by jerryz123