riscv-boom/riscv-boom #531: questions about ringboom

siyy123, 2021-03-13:
I'm recently doing some research about very big core and I've found ringboom. But I have some question about it.
1.Is ringboom still maintained? In which branch contains the latest ringboom source code?
2.If there are some similar capacities in the main branch?
3.If I use ringboom which version of chipyard should I use?
I'd appreciate it if you could give me clear answers soon.

jerryz123, 2021-03-24:
1. Ringboom is not actively maintained. Many of the features of ring-boom are slowly getting tested and will be merged into the main release.

2. The main branch and ringboom have some similar optimizations. I currently have another development branch that has more of the ring-boom optimizations verified, but I have not had time to merge it yet.

siyy123, 2021-03-25:
Ok. Thanks for your clear answer

closed 2021-03-25 by siyy123