riscv-boom/riscv-boom #533: SonicBoom chickenCSR

sabbaghm, 2021-03-17:
**Type of issue**: question

Is the Chicken CSR's address still valid for SonicBoom?

`0x7c1` as mentioned here.
https://github.com/riscv-boom/riscv-boom/blob/840d4e51ce0c4319f0ae8243487a137fd9745963/docs/sections/debugging.rst

`csrwi 0x7c1, 0x8` or `csrwi 0x7c1, 0x0`  produces an illegal instruction during run-time on my compilation (1-wide, SmallCore, SonicBoom). Could you please provide your sample CSR write code (e.g., asm or in-line asm for C) for the chickenCSR that runs successfully on SonicBoom? Thanks.

abejgonzalez, 2021-03-17:
According to the RISC-V privileged spec. `0x700` to `0x77f` are machine mode CSRs. Are you making sure to write/read the CSR in machine mode? If you want, you can try to switch the CSR to be a user/supervisor/hypervisor (look at Table 2.1 in the priv. spec)

sabbaghm, 2021-03-17:
> According to the RISC-V privileged spec. `0x700` to `0x77f` are machine mode CSRs. Are you making sure to write/read the CSR in machine mode? If you want, you can try to switch the CSR to be a user/supervisor/hypervisor (look at Table 2.1 in the priv. spec)

I see. No, I was running it in supervisor mode. I will then change the address to `0x5c1` in Rocket/BOOM Scala sources (to be in `0x5C0-0x5FF`). Could you please point me to the required steps If I were to run it only in machine mode on SonicBoom? thanks.

abejgonzalez, 2021-03-18:
Depends on what you are running. If you are running something like Linux, you would need to do write the chicken bit with an SBI call (which is in turn called by a Linux syscall). You can look at `riscv-pk`'s `mcall_console_putchar`, Linux's `sbi_console_putchar`. In the end, unless you need the CSR to be machine mode... I would just recommend changing it to user/supervisor/hypervisor mode.

sabbaghm, 2021-03-18:
> Depends on what you are running. If you are running something like Linux, you would need to do write the chicken bit with an SBI call (which is in turn called by a Linux syscall). You can look at [...]

Thanks a lot! Indeed helpful. 
Yes, I am running Linux (debian64). Right, I think for testing purposes "non-machine mode" is what we want, but for the final implementation, a machine-mode ChickenCSR (as it is) makes more sense (for obvious security reasons). I was thinking of placing the ChickenCSR as part of `bootrom.S` (rocket-chip) or can we build something like "BIOS"/coreboot for SonicBoom pre-OS boot settings. I am not sure though if it is a viable solution. Thanks!