riscv-boom/riscv-boom #534: When branch prediction error occurs, RoCC interrupt is ignored

vinackylee, 2021-03-23:
**Type of issue**: bug report

We want to send an interrupt to BOOM core through RoCC. First, we set CSR registers(mie: 1888, mstatus: 1e088, mtvec: 80001048). Then a RoCC interrupt is 1 for only one cycle. Finally, we can't find the interrupt and the program don't jump to 80001048.

Through the analysis of our program and BOOM, we found the reason for the above phenomenon. Our program has many branchs. We can see the interrupt signal arrives at CSRFile moudle and is send to decode stage. Befor it is send to rename stage a branch error occur from branch unit. Then the instruction with interrupt signal is cancelled because of the branch error.

```scala
//core.scala
  val bru_pc_req  = Wire(Decoupled(UInt(log2Ceil(ftqSz).W)))
  val xcpt_pc_req = Wire(Decoupled(UInt(log2Ceil(ftqSz).W)))

  val ftq_arb = Module(new Arbiter(UInt(log2Ceil(ftqSz).W), 2))

  ftq_arb.io.in(0) <> bru_pc_req
  ftq_arb.io.in(1) <> xcpt_pc_req

  // Hookup FTQ
  io.ifu.get_pc.ftq_idx := ftq_arb.io.out.bits
  ftq_arb.io.out.ready  := true.B

  // Branch Unit Requests
  bru_pc_req.valid := RegNext(iss_valids(brunit_idx))
  bru_pc_req.bits  := RegNext(iss_uops(brunit_idx).ftq_idx)
  exe_units(brunit_idx).io.get_ftq_pc := DontCare
  exe_units(brunit_idx).io.get_ftq_pc.fetch_pc := RegNext(io.ifu.get_pc.fetch_pc)
  exe_units(brunit_idx).io.get_ftq_pc.next_val := RegNext(io.ifu.get_pc.next_val)
  exe_units(brunit_idx).io.get_ftq_pc.next_pc  := RegNext(io.ifu.get_pc.next_pc)

  // Frontend Exception Requests
  val xcpt_idx = PriorityEncoder(dec_xcpts)
  xcpt_pc_req.valid    := dec_xcpts.reduce(_||_)
  xcpt_pc_req.bits     := dec_uops(xcpt_idx).ftq_idx
  rob.io.xcpt_fetch_pc := RegEnable(io.ifu.get_pc.fetch_pc, dis_ready)
```

vinackylee, 2021-03-23:
If we delete the branch instruction in the program and add the NOP instruction to wait for the interrupt, the program can jump to 80001048.

jerryz123, 2021-03-23:
You must hold the interrupt signal high for multiple cycles. There is no guarantee that the core will be ready to handle the interrupt on the exact cycle your RoCC unit is setting it high.

vinackylee, 2021-03-23:
If we hold the interrupt signal high for n+1 cycles(n is the pipeline depth), can we guarantee the core get the interrupt?

jerryz123, 2021-03-23:
No. The core may be in the process of handling some other interrupt.

You should hold io.interrupt high in your rocc acccelerator indefinitely. Then, in your interrupt handler, send a special rocc instruction which clears the interrupt.

closed 2021-09-29 by jerryz123