riscv-boom/riscv-boom #536: Question about issue unit

lxu28973, 2021-03-27:
**Type of issue**: question

I read the code in file issue-unit.scala, and have a question about the code on line 125.
Here is the code between line 125 and 142:
```scala
    if (iqType == IQT_MEM.litValue || iqType == IQT_INT.litValue) {
      // For StoreAddrGen for Int, or AMOAddrGen, we go to addr gen state
      when ((io.dis_uops(w).bits.uopc === uopSTA && io.dis_uops(w).bits.lrs2_rtype === RT_FIX) ||
             io.dis_uops(w).bits.uopc === uopAMO_AG) {
        dis_uops(w).iw_state := s_valid_2
        // For store addr gen for FP, rs2 is the FP register, and we don't wait for that here
      } .elsewhen (io.dis_uops(w).bits.uopc === uopSTA && io.dis_uops(w).bits.lrs2_rtype =/= RT_FIX) {
        dis_uops(w).lrs2_rtype := RT_X
        dis_uops(w).prs2_busy  := false.B
      }
      dis_uops(w).prs3_busy := false.B
    } else if (iqType == IQT_FP.litValue) {
      // FP "StoreAddrGen" is really storeDataGen, and rs1 is the integer address register
      when (io.dis_uops(w).bits.uopc === uopSTA) {
        dis_uops(w).lrs1_rtype := RT_X
        dis_uops(w).prs1_busy  := false.B
      }
    }
```
Why we also check if uopc equal to uopSTA or uopAMO_AG(line 3 and 4) when "iqType == IQT_INT.litValue"(line 1)? In other words, why these code:
```scala
      when ((io.dis_uops(w).bits.uopc === uopSTA && io.dis_uops(w).bits.lrs2_rtype === RT_FIX) ||
             io.dis_uops(w).bits.uopc === uopAMO_AG) {
        dis_uops(w).iw_state := s_valid_2
        // For store addr gen for FP, rs2 is the FP register, and we don't wait for that here
      } .elsewhen (io.dis_uops(w).bits.uopc === uopSTA && io.dis_uops(w).bits.lrs2_rtype =/= RT_FIX) {
        dis_uops(w).lrs2_rtype := RT_X
        dis_uops(w).prs2_busy  := false.B
      }
```
also needed when iqType == IQT_INT.litValue. According to the decode table, all instuctions whose uop equal to uopSTA or uopAMO_AG will be dispatched to MEM issue queue.

jerryz123, 2021-03-27:
Right. I believe this code only needs to apply when `iqType == IQT_MEM`

lxu28973, 2021-03-28:
Thanks for your reply.

closed 2021-03-28 by lxu28973