riscv-boom/riscv-boom #541: loop predictor bug for loops with small body

gxflying, 2021-04-16:
**Type of issue**: bug report | feature request | question | other enhancement
bug report

**Impact**: no rtl change | rtl refactoring | new rtl | unknown
rtl refactoring

**Development Phase**: request | proposal
request
**Other information**

when i test the loop predictor by asembly code below where the loop body can be putted in one fetchboundle , I found that the loop predictor actrual updates the wrong 
value into the s_cnt, this is probablu because of the delay of the pipeline, 
code :
main:
li t1, 40
1:
li t0 , 100
2:
addw t1, t1, -1
bneq t1, 2b
li t1 ,40
addw t0, t0, -1
bneq t0, 2b
j 1b  # loop forever fot test

run the assembly code above, can easily reproduce the bug
**What is the current behavior?**
the valuses of the s_cnt in the entry seems to be two interleaved sequences:
f3_entry_s_cnt :  0 | 49 | 1 | 50 | 2 | 51 | 3 | 52 | 4 | 52 |  ...
[image]

**What is the expected behavior?**
the s_cnt should be a successive increased sequence
**Please tell us about your environment:**

**What is the use case for changing the behavior?**
main:
li t1, 40
1:
li t0 , 100
2:
addw t1, t1, -1
bneq t1, 2b
li t1 ,40
addw t0, t0, -1
bneq t0, 2b
j 1b  # loop forever fot test
[image]

jerryz123, 2021-04-23:
Thanks. Could you post the binary which demonstrates this case?

gxflying, 2021-04-26:
actually , this is all the code for the test,  and I sorry the bin file is on a server from which I can not take things out.
the disassembly code is in the pictrue bellow
```
main:
li t1, 40
1:
li t0 , 100
2:
addw t1, t1, -1
bneq t1, 2b
li t1 ,40
addw t0, t0, -1
bneq t0, 2b
j 1b # loop forever fot test
```
[image]