riscv-boom/riscv-boom #542: Change the clock of SOC

Sjf-bianhuakuai, 2021-04-23:
Hi,
   I'm study the SOC of the BoomV3 and find that all clocks are same.So I want to add a clock to it from IO.I find that I should change the code of {ChipTop.scala、AsyncCrossing.scala、MultiClock.scala、AsyncQueue.scala and so on}.
 Now I haven't done it.I'd like to confirm if my research direction is wrong。Looking forward to your reply, thank you！
---

jerryz123, 2021-04-23:
Multiclock configurations are supported on the latest Chipyard `dev` branch. It is possible to put the Tiles on different clock domain than the rest of the system. For such a system, no RTL needs to be modified. There are existing configurations for multiclock systems which can be adapted to specify multiclock SonicBOOM-based systems.

Everything within the SonicBOOM tile (frontend/core/lsu/dcache) must run in the same clock frequency.

Sjf-bianhuakuai, 2021-04-24:
Last Chipyard dev branch? Sorry I don't understand what you mean ,Maybe I didn’t express it clearly.
I found that the BoomV3 has a input clock from IO and then this clock is copied into six by the systemBus to FrontBus 、L2、MemoryBus 、ControlBus 、PeripheralsBus and Tilecore  respectively. So these six Module of Chipyard are used the same clock.  Now I want to change them。But this can be  easied done in BoomV2.
The interface of BoomV3 is not same as the BoomV2. So  I want to know if I need to modify a lot of files(such as what I mentioned above) to achieve this goal .

Sjf-bianhuakuai, 2021-04-25:
Hi,
   I'm sorry that  I misunderstood you and the difference between BoomV2 and BoomV3.
  Do you mean that  I only need to  change the  Multiclock configurations to get different clocks which used in different modules of the Chipyard SOC? And don't  have to modify many files such as ChipTop.scala、AsyncCrossing.scala、MultiClock.scala、AsyncQueue.scala and so on? 
  Thank you.

Sjf-bianhuakuai, 2021-05-06:
Hi, 
    I download the last chipyard/generators ,but the simulation core was not generated successfully. I found that the common.mk is different and cva6 added ,ariane removed .Do I need to update all the files of chipyard,such as generators 、project、scripts an so on?

closed 2021-05-11 by Sjf-bianhuakuai