riscv-boom/riscv-boom #545: question on the ready signal connects between ifu and decoder

gxflying, 2021-05-06:
Hi sir, I am learn about the soincboom code, and I  found that the handshake "ready" signal between ifu & decoder is confused and may wrong. see the code below,  assumed the coreWidth=4
the ifu will dequeue 4 instructions at once when avilable and "dec_ready" asserts  
but  the signel dec_fire.last only indicates that the last instruction in the packets, not all the 4 instructons are ready !!
and in my simulation （the picture blow）,  
**dec_fire_0=0, dec_fire_1=0, dec_fire_2=1, dec_fire_3=1 but the dec_ready = 1 ???**    does this really ready fir dec_fire_0 ?

https://github.com/riscv-boom/riscv-boom/blob/e252e797c21aa3bf860cb9d67c6009ef00c5916f/src/main/scala/exu/core.scala#L576
https://github.com/riscv-boom/riscv-boom/blob/e252e797c21aa3bf860cb9d67c6009ef00c5916f/src/main/scala/exu/core.scala#L579
[image]

jerryz123, 2021-05-06:
If the last instruction in a fetchpacket is ready to move past decode (dec_fire_3), then due to the requirement for in-order decode/rename/dispatch, the previous instructions must either also be ready, or moved past decode in an earlier cycle.

gxflying, 2021-05-06:
yeah, I see , thanks.
and the ifu.fetchpacket may pass partial of packet instructions to decoder without  increasing the head pointer in fetchbuffer when will_hit_tail asserts.

closed 2021-05-06 by gxflying