riscv-boom/riscv-boom #549: TLMonitor Assertion failure while running baremetal on BOOM

JaewonHur, 2021-06-21:
**Type of issue**: bug report

**Impact**: unknown

**Other information**
I'm using SmallBoomConfig of riscv-boom v3.0.0 (without any modification).

While running a bare-metal binary (which I modified from the SpectreAttack repo),
The simulation occasionally failed due to the assertion as below.
> [235] %Error: chipyard.TestHarness.SmallBoomConfig.top.v:312740: Assertion failed in 
> TOP.TestHarness.dut.system.boom_tile.buffer.TLMonitor
> %Error: /home/jwhur/Research/SpecDoctor/chipyard/sims/verilator/generated
> src/chipyard.TestHarness.SmallBoomConfig/chipyard.TestHarness.SmallBoomConfig.top.v:312740: Verilog $stop
> Aborting...

After checking the waveform, It seems the request from the `HellaCache` is illegally generated.
Here's the binary, dump, corresponding waveform, and commit log.

[bug.zip](https://github.com/riscv-boom/riscv-boom/files/6684740/bug.zip)

jerryz123, 2021-06-21:
Can you upload a zip of the entire VPD?

JaewonHur, 2021-06-21:
I could not upload the entire vcd file since it was too big.

JaewonHur, 2021-06-21:
I'm sorry for taking your time.
It was my mistake, I had modified the binary to return to the supervisor mode.

Sorry for the confusion.

jerryz123, 2021-06-22:
This might still be a bug. I would expect a access fault in this case