riscv-boom/riscv-boom #552: Store miss which is accepted by MSHR will response to lsu twice？

ddxoxpp, 2021-07-15:
**Type of issue**:  question
   The code of dcache in line 732 is listed below. The store miss accepted by mshrs will responsed to lsu when it was fired to mshrs. And it will response to lsu for the second time when replay? What will happen when the second response to the same store changes the younger store's succeed reg in the same stq entry ?
 `  val s2_send_resp = widthMap(w => (RegNext(s1_send_resp_or_nack(w)) && !s2_nack(w) &&
                      (s2_hit(w) || (mshrs.io.req(w).fire() && isWrite(s2_req(w).uop.mem_cmd) && !isRead(s2_req(w).uop.mem_cmd))))) `
[https://github.com/riscv-boom/riscv-boom/blob/master/src/main/scala/lsu/dcache.scala#:~:text=val%20s2_send_resp%20%3D%20widthMap,w).uop.mem_cmd)))))](url)

jerryz123, 2021-07-16:
`s0_send_resp_or_nack` should only be high for loads.

ddxoxpp, 2021-07-16:
> `s0_send_resp_or_nack` should only be high for loads.

I didn't notice that this signal changes in replay. 
Thanks for your answer!

ddxoxpp, 2021-07-16:
And could you tell me how the lsu make sure that the older load must be executed before the younger store which has the same address? I just see the younger load would use forward and s1_kill to be executed after the old same address store. @jerryz123

jerryz123, 2021-07-16:
Stores don't execute until commit, at which point all older loads would have executed.

ddxoxpp, 2021-07-17:
> Stores don't execute until commit, at which point all older loads would have executed.

But AMOs execute before commit.

jerryz123, 2021-07-17:
AMOs don't execute until they are at the commit point in BOOM. Allowing AMOs to execute ahead of commit would make maintaining the memory model more challenging.

ddxoxpp, 2021-07-17:
> AMOs don't execute until they are at the commit point in BOOM. Allowing AMOs to execute ahead of commit would make maintaining the memory model more challenging.

Thanks for your reply. I just have read lsu and dcache code and can't found where and which code to make sure that "AMOs don't execute until they are at the commit point in BOOM". Could you please point it out for me, thanks.

jerryz123, 2021-07-17:
Sorry on phone can't show you the code.
But if you just search for amo in lsu.scala you'll find lots of special
case code for amos

ddxoxpp, 2021-07-17:
I have searched amo in lsu and didn‘t get any signals that indicate AMO is fired when it’s at the commit point. So I thought it is achieved in other module, such as when it‘s at the commit head , the core will send valid addr and data to AMO in lsu, I guess?
'  val can_fire_store_commit  = widthMap(w =>
                               ( stq_commit_e.valid                           &&
                                !stq_commit_e.bits.uop.is_fence               &&
                                !mem_xcpt_valid                               &&
                                !stq_commit_e.bits.uop.exception              &&
                                (w == 0).B                                    &&
                                (stq_commit_e.bits.committed || ( stq_commit_e.bits.uop.is_amo      &&
                                                                  stq_commit_e.bits.addr.valid      &&
                                                                 !stq_commit_e.bits.addr_is_virtual &&
                                                                  stq_commit_e.bits.data.valid))))
'

closed 2021-09-09 by ddxoxpp