riscv-boom/riscv-boom #555: What if bpd_end_idx has exceeded bpd_end_idx when starting the check of repair end  in Line 259-266?

ddxoxpp, 2021-08-23:
https://github.com/riscv-boom/riscv-boom/blob/e252e797c21aa3bf860cb9d67c6009ef00c5916f/src/main/scala/ifu/fetch-target-queue.scala#L259-L266
I notice that FTQ start checking the repair end after the bpd_repair_idx +2, so what if bpd_end_idx has exceeded bpd_end_idx, which will send bpdupdate request using all FTQ entrys  including those invalid entrys？
And there are some else questions confusing me：
When handling mispredict repair (which takes some cycle to finish), what if the redirect generated by rob.flush  happens and interrupt the mispredict repair?

jerryz123, 2021-08-23:
> what if bpd_end_idx has exceeded bpd_end_idx

What do you mean here?

> When handling mispredict repair (which takes some cycle to finish), what if the redirect generated by rob.flush happens and interrupt the mispredict repair?

The FTQ aborts the current repair and start repairing from the new mispredict.

ddxoxpp, 2021-08-23:
> > what if bpd_end_idx has exceeded bpd_end_idx
> 
> What do you mean here?
https://github.com/riscv-boom/riscv-boom/blob/e252e797c21aa3bf860cb9d67c6009ef00c5916f/src/main/scala/ifu/fetch-target-queue.scala#L245-L266
```WrapInc(bpd_repair_idx, num_entries) === bpd_end_idx``` checking begins after ```WrapInc(bpd_repair_idx,num_entries)``` executing twice , I means  if ```bpd_repair_idx``` is  “bigger” than ```bpd_end_idx``` when beigning ```WrapInc(bpd_repair_idx, num_entries) === bpd_end_idx``` checking,  invalid FTQ entrys will generate bpdupdate message.  I think "bigger" is not exact description in circular queue，so I said “exceed”.

jerryz123, 2021-09-03:
Hmm, that's a good observation, and it would certainly be a bug. I believe in practice this is unlikely, as it is unlikely for there to many invalid FTQ entries immediately after the FTQ entry of a branch mispredict, as 1 FTQ entry is allocated per cycle in steady state, and it takes several cycles between fetch and branch resolution

ddxoxpp, 2021-09-08:
> Hmm, that's a good observation, and it would certainly be a bug. I believe in practice this is unlikely, as it is unlikely for there to many invalid FTQ entries immediately after the FTQ entry of a [...]

What if two mispredictions in two consecutive ‘’fetchpacket" happen back-to-back?

jerryz123, 2021-09-29:
That could cause the issue.