riscv-boom/riscv-boom #558: Is there any possibility that ```RegNext(rob.io.flush.valid)``` and ```brupdate.b2.mispredict``` are valid in the same cycle, which makes the mispredict FTQ entry's information writen into the redirect FTQ entry?

ddxoxpp, 2021-09-08:
https://github.com/riscv-boom/riscv-boom/blob/e252e797c21aa3bf860cb9d67c6009ef00c5916f/src/main/scala/ifu/fetch-target-queue.scala#L314-L338
```RegNext(rob.io.flush.valid)``` makes ```io.redirect.valid``` in FTQ is valid, which leads to writing rob flush redirect FTQ entry to ```prev_entry``` and FTQ entry ram.
But if ```brupdate.b2.mispredict``` is valid in the same cycle, which  changes the rob flush redirect FTQ entry's infomation to mispredict FTQ entry's information.
Is there any possibility that ```RegNext(rob.io.flush.valid)``` and ```brupdate.b2.mispredict``` are valid in the same cycle, which makes the mispredict FTQ entry's information writen into the ```RegNext(rob.io.flush.valid)``` redirect FTQ entry?

jerryz123, 2021-09-08:
I believe this should be impossible. The flush signal would have killed the in-flight branch in the previous cycle, so it would be impossible for the branch mispredict signal to also be high.

ddxoxpp, 2021-09-09:
> I believe this should be impossible. The flush signal would have killed the in-flight branch in the previous cycle, so it would be impossible for the branch mispredict signal to also be high.

https://github.com/riscv-boom/riscv-boom/blob/e252e797c21aa3bf860cb9d67c6009ef00c5916f/src/main/scala/exu/core.scala#L192-L195
But the ```rob.io.flush.valid``` doesn't prevent the last cycle's ```exe_units.alu_units.io.brinfo``` writing into the ```brinfo``` reg, which results in ```brupdate.b2.mispredict``` is valid in the same cycle when the ``` RegNext(rob.io.flush.valid)``` is valid.

jerryz123, 2021-09-09:
The valid signal is gated off by the flush, which suppresses the mispredict signal the next cycle.

JaewonHur, 2021-11-01:
Hi, it seems this can actually happen.
Here is the testcase and the vcd file.

Both `io_redirect_valid` and `io_brupdate_b2_mispredict` signals of `ftq` are simultaneously asserted at 150887 ps. 
But the corresponding indices (`io_redirect_bits` and `io_brupdate_b2_uop_ftq_idx`) are different, thus updating the wrong ftq entry. 

I tested the case on boom commit: riscv-boom/riscv-boom@d77c2c3
You can reproduce the vcd file with following command.
`./simulator-chipyard-SmallBoomConfig-debug -x 70000 --vcd=bug.vcd --seed=0 bug.riscv`

[bug.zip](https://github.com/riscv-boom/riscv-boom/files/7451653/bug.zip)

jerryz123, 2021-11-01:
I see. This is just a performance bug, then, not a functionality bug? The incorrectly updated entry should only cause some incorrect branch predictor updates.