riscv-boom/riscv-boom #563: question about TAGE IUM

Lingrui98, 2021-09-22:
**Type of issue**: bug report | feature request | question | other enhancement
question

Q: I noticed there was a commit b836971a2911fcd7c12b0eade84620b7df2b6f2c which removed IUM from TAGE and claimed that wrbypass was better. IUM is proposed to deal with the impact of delayed update, however a wrbypass does not seem to manage it. Is there any more evidence on that decision?

jerryz123, 2021-09-22:
The issue I was attempting to solve at the time was the TAGE counters not being incremented properly in tightly looping code.
The IUM-solution fixed the counter value when it is read, while the wrbypass solution fixes the counter value when it is written.
While it is true that IUM addresses problems from delayed update in TAGE, it also caused some degenerate and confusing behavior, which is why I removed it.

Lingrui98, 2021-09-23:
It seems that you implemented the IUM in TageTable, and store counters in IUM without updating it. I think that's why it caused the counter value problem. According to the original paper (and code), the IUM stores only the direction, and does not modify the predicted counters. Have you tried the original version of IUM?

jerryz123, 2021-09-29:
Yeah now that you mention it, updating the IUM counters in the IUM directly would have been the better implementations.

Lingrui98, 2021-09-30:
Thanks for your reply!

closed 2021-09-30 by Lingrui98