riscv-boom/riscv-boom #573: 'C' channel ProbeAck carries invalid source

Yujie-Cui, 2021-10-25:
We built Boom based on https://docs.boom-core.org/en/latest/. And it can be compiled successfully, but the problem of 
```
Assertion failed: 'C' channel ProbeAckData carries invalid source ID (conCrossingHelper.scala:61:80)
    at Monitor.scala:42 assert(cond, message)
[1592000] %Error: chipyard.TestHarness.MediumBoomConfig.top.v:426777: Assled in TOP.TestHarness.chiptop.system.tile_prci_domain.buffer_1.monitor
%Error:  chipyard/sims/verilator/generated-src/chipyard.TestdiumBoomConfig/chipyard.TestHarness.MediumBoomConfig.top.v:426777: Verilo
Aborting...

```
 appeared during execution.  why is this?

Thanks!

jerryz123, 2021-11-02:
What commit are you using?

jerryz123, 2021-11-07:
Duplicate of #459

closed 2021-11-07 by jerryz123