riscv-boom/riscv-boom #576: Will the error-predicted jalr instructions also modify the BTB status in BOOM?

hz1490919302, 2021-11-06:
Will the error-predicted jalr instructions also modify the BTB entries in BOOM, like in other processors?
Does BOOM have any unique design for BTB so that the error prediction jalr instruction will not update the BTB entry?

jerryz123, 2021-11-06:
In BOOM BTB updates from JALR should not reach the BTB until commit, so a mispredicted JALR should not update the BTB.

hz1490919302, 2021-11-06:
Thanks, that's very useful. So does RAS also update after the call instruction is committed? Or does RAS update once it meets a call instruction in fetch stage?

jerryz123, 2021-11-06:
RAS is updated speculatively in fetch. There is a repair mechanism to reduce RAS corruption caused by misspeculated calls.

hz1490919302, 2021-11-06:
Is this repair mechanism optional? Or what code can I modify to get a RAS without a repair mechanism.(Because I need to finish my academic thesis on RISC-V. I need to implement some Spectre attacks using BTB or RAS, but the RISC-V BOOM is designed to be very secure, which makes it difficult to complete these attacks.)

hz1490919302, 2021-11-06:
[image]
Hi, maybe I found the switch for the RAS repair mechanism you mentioned,, I'll try to do a Spectre attack test on RAS side channel, if I can, I'll submit my attack code and the defense measures to BOOM in my paper.

jerryz123, 2021-11-07:
Great! You can PR the attack to boom-attacks if you would like.

closed 2021-11-07 by jerryz123