riscv-boom/riscv-boom #609: Thanks for the question!

RutvikC, 2022-07-19:
Thanks for the question!

First of all, any suggestions to clarify the documentation is always helpful. Sorry it didn't help to much in this case.

1. Which events does BOOM track? 

Here are the snippets of code that define the events.

https://github.com/riscv-boom/riscv-boom/blob/f4968d770b6d0a461f81d26956a141e4300b3e45/src/main/scala/exu/core.scala#L190-L195
https://github.com/riscv-boom/riscv-boom/blob/f4968d770b6d0a461f81d26956a141e4300b3e45/src/main/scala/exu/core.scala#L221-L228
https://github.com/riscv-boom/riscv-boom/blob/f4968d770b6d0a461f81d26956a141e4300b3e45/src/main/scala/exu/core.scala#L245-L251

They tracks things such as branch prediction, ROB flushes, I/D$ misses, etc. As for the numbering/ordering of the events, we reference the U54 manual as an example on how to access events (if I remember correctly they map to the Rocket code). The events that they have (and what Rocket has) will be different from BOOM. You need to look at the `exu/core.scala` to see the signals that we chose and access them using the EventSet and EventBit that is relevant.

Note: If you want to track your own event, feel free to modify the source and add your own.

2. How many counters do we support?

We use the same infrastructure to support counters as Rocket. So if I am not mistaken we support 32 (including cycle, time, instret counters).

3. A library to use the counters (configuring and reading them is kinda ugly)?

What I end up using for my own projects is this repository: https://github.com/abejgonzalez/riscv-code-constructor. It includes a header file https://github.com/abejgonzalez/riscv-code-constructor/blob/master/inc/hpm-util.h and example on how to use it https://github.com/abejgonzalez/riscv-code-constructor/blob/master/src/hpmCounters.c. Otherwise, `riscv-tests` uses `setStats` to read the counters.

https://github.com/riscv/riscv-tests/blob/fbf5f3a2589c61d34569524dbf353beda0b6b4de/benchmarks/common/syscalls.c#L40-L54

Finally, Chris Celio created a program to help reading counters also: https://github.com/ccelio/riscv-hpmcounters

Otherwise, as far as I know, there is no "de-facto" standard for reading out counters. So you have to hack something together or use the code found in one of these sources (or somewhere else that I don't know of).

Please let me know if you have any other questions! I can also update the docs with this information if that was unclear.

_Originally posted by @abejgonzalez in https://github.com/riscv-boom/riscv-boom/issues/265#issuecomment-492818596_

RutvikC, 2022-07-19:
Regarding this reply, I tried following step 3.3 mentioned here (https://inst.eecs.berkeley.edu/~cs152/sp21/assignments/sp21/lab2.pdf) and compiled the **SmallBoomConfig** and ran the hpmCounter.riscv file on Verilator.

But my main concern is the performance counter output when using the above mentioned BOOM config. This is the output that I am getting:
[image]

Why am I not getting an output other than 0 for the events that are setup in the hpmCounter.c code?

RutvikC, 2022-07-23:
Had to change the **_nPerfCounters_** value where the **_WithNSmallBooms_** class was defined in the _config-mixins.scala_

closed 2022-07-23 by RutvikC

Rai0Catodic0, 2023-08-04:
Hi. 
I'm trying to obtain the number of branch missprediction through HPC and I'm foloowing the steps listed above in the BOOM from Chipyard 1.10.0.

the lines in https://github.com/riscv-boom/riscv-boom/blob/f4968d770b6d0a461f81d26956a141e4300b3e45/src/main/scala/exu/core.scala#L221-L228 are comented and when I uncomment those lines and try to compile BOOM or Rocket on verilator I get the following:

[image]

EternitYjl, 2025-01-22:
[image]
I had the same issue. So did you solve the problem? If you have any good solution, please tell me. Thank you!

EternitYjl, 2025-01-22:
> Hi. I'm trying to obtain the number of branch missprediction through HPC and I'm foloowing the steps listed above in the BOOM from Chipyard 1.10.0.
> 
> the lines in
> 
> [riscv-boom/src/main/scala/exu/core.scala](https://github.com/riscv-boom/riscv-boom/blob/f4968d770b6d0a461f81d26956a141e4300b3e45/src/main/scala/exu/core.scala#L221-L228)
> 
> Lines 221 to 228 in [f4968d7](/riscv-boom/riscv-boom/commit/f4968d770b6d0a461f81d26956a141e4300b3e45)
> 
>  new freechips.rocketchip.rocket.EventSet((mask, hits) => (mask & hits).orR, Seq( 
>    ("I$ blocked",                        () => icache_blocked), 
>    ("nop",                               () => false.B), 
>    ("branch misprediction",              () => br_unit.brinfo.mispredict), 
>    ("control-flow target misprediction", () => br_unit.brinfo.mispredict && 
>                                                br_unit.brinfo.is_jr), 
>    ("flush",                             () => rob.io.flush.valid), 
>    ("branch resolved",                   () => br_unit.brinfo.valid))), 
> are comented and when I uncomment those lines and try to compile BOOM or Rocket on verilator I get the following error:
> <img alt="Captura de tela 2023-08-04 164715" width="781" src="https://private-user-images.githubusercontent.com/47091917/258515976-22ec3e22-caaa-4317-b092-6ac9c1109eb4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzc1MTg1NDQsIm5iZiI6MTczNzUxODI0NCwicGF0aCI6Ii80NzA5MTkxNy8yNTg1MTU5NzYtMjJlYzNlMjItY2FhYS00MzE3LWIwOTItNmFjOWMxMTA5ZWI0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTIyVDAzNTcyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkwZjMyNGFjOGVmOGNkMjk5MDQ0Njg4ZWNjMWM5NTM4Yzg4ZWIxNjFiYTU0YjQyYmRjOTQ2YmE0YjdhMTA5ZjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.9Xe53DnsNKcAYA-5VHWs4lEnmIS3h4aG2Ltoy7chLY8">

I had the same issue. So did you solve the problem? If you have any good solution, please tell me. Thank you!