riscv-boom/riscv-boom #612: BOOM v3 fails to run pk in the Chipyard environment.

hahaha142142, 2022-09-03:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**Other information**

In Chipyard, "SmallBoomConfig" and "MegaBoomConfig" both cannot run pk successfully, but "RocketConfig" can.
Command 1:
`./simulator-chipyard-MegaBoomConfig-debug -V pk`
Result 1:
`using random seed 1662202065`
`This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.`
`Listening on port 45253`
`[UART] UART0 is here (stdin/stdout).`
`UART TX (62): b`
`*** PASSED *** Completed after 610079 cycles`
Command 2:
`./simulator-chipyard-SmallBoomConfig -V pk`
Result 2:
`using random seed 1662208359`
`This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.`
`Listening on port 36555`
`[UART] UART0 is here (stdin/stdout).`
`UART TX (62): b`
`bUART TX (62): b`
`*** PASSED *** Completed after 659803 cycles`
Command 3:
`./simulator-chipyard-RocketConfig pk`
Result 3:
`This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.`
`Listening on port 43639`
`[UART] UART0 is here (stdin/stdout).`
`bbl loader`
`tell me what ELF to load!`
`*** FAILED *** (code = -1, seed 1662208269) after 794761 cycles`
In fact, the result 3 of "RocketConfig" is the correct result, which is consistent with the "Spike" result.
Command "Spike":
`spike pk`
Result "Spike":
`bbl loader`
`tell me what ELF to load!`
By the way, I'm not sure if it is a bug. Maybe I miss something.

**if the current behavior is a bug, please provide the following steps to reproduce the problem**
Using BOOM core in Chipyard to run pk.

**What is the current behavior?**
Having inconsistent result with "Spike".

**What is the expected behavior?**
Having consistent result with "Spike".

**Please tell us about your environment:**

Chipyard 1.6.2 with [boom @ ad64c54](https://github.com/riscv-boom/riscv-boom/tree/ad64c5419151e5e886daee7084d8399713b46b4b).

**What is the use case for changing the behavior?**
The pk case.

jerryz123, 2022-09-03:
Does the issue appear with Chipyard 1.7.1?

hahaha142142, 2022-09-05:
> Does the issue appear with Chipyard 1.7.1?

I build a new Chipyard 1.7.1 environment, and find what results in the issue.
If I build toolchains in the general way, which uses "rv64gc" to compile pk, there exists no problem in regarding to BOOM core.
However, if I build toolchains in the custom way, which uses "rv64ia" to compile pk, there exists the problem mentioned above in regarding to BOOM core (while no problem in regarding to Rocket core).
So, maybe BOOM core has something unsuited to pk compiled with "rv64ia"?