riscv-boom/riscv-boom #615: Are neighboring loads handled together in BOOM?

sammy17, 2022-09-22:
**Type of issue**: question 

I noticed that in BOOM when there are two loads a few instructions apart, the earlier load is delayed. 
I wanted to check whether this behavior is expected in BOOM due to some feature. 

Are nearby loads handled together in BOOM?

jerryz123, 2022-09-22:
Loads are not coalesced, if that is what you are asking.
What do you mean "the earlier load is delayed". That should not happen.

ccelio, 2022-09-22:
We would have to see the instruction stream, and better yet, a view of the pipe trace to diagnose accurately. For example, if your second load depends on the value of the first load to compute its address, you can't execute those back-to-back.

sammy17, 2022-09-22:
There are no data dependencies between the loads. 
Following is the instruction sequence I am using:
```
seq1:
auipc   a7, 0x20
addi    a7, a7, -804
lw      a1, -28(a7)    # address = 0x80020104
nop
nop
nop
nop
nop
nop
nop
auipc   s0, 0x30
addi    s0, s0, -656
nop
```

This is my initial instruction sequence. Then I replace the last nop instruction with another lw. 

```
seq2:
auipc   a7, 0x20
addi    a7, a7, -804
lw      a1, -28(a7)   # address = 0x80020104
nop
nop
nop
nop
nop
nop
nop
auipc   s0, 0x30
addi    s0, s0, -656
lw      t5, 140(s0)      # address = 0x8003024c
```
By comparing the commit times of instructions in seq1 and seq2, I see that in seq2, the first lw instruction is delayed by additional 8 cycles compared to seq1. 
This is what I meant as "the earlier load is delayed".

The only reason I could think of this happening was if BOOM coalesced the loads, it should not be a delay due to cache activity as addresses are not related. 
Let me know if you could think of any reason why this is happening. 

I print the internal RTL signals to check the commit times of instructions. 
Also, is there a way to get a commit trace from the latest BOOM so that I can verify this behavior with that as well?

Attached binaries: test_boom_lw_nop.elf has the seq1, test_boom_lw_lw.elf has the seq2. 
[tests.zip](https://github.com/riscv-boom/riscv-boom/files/9628926/tests.zip)