riscv-boom/riscv-boom #624: MediumBoomConfig does not boot Linux because of a hazard

Schmitz48, 2022-12-19:
**Type of issue**: bug report 

**Impact**: unknown

**Development Phase**: request 

**Other information**

As mentioned here [https://github.com/ucb-bar/chipyard/issues/1279](issue) the core hangs during the Linux boot up. The issue first occured on the VCU118 FPGA but it can be reproduced with simulation of the boot up process.  I traced it down to the following scenario: An sfence invalidates the TLB entry with the tag 0x4efeefe for e.g., virtual address 0xcefeefe008. After this there is a store (ST1) to vaddr 0xe00167777f0 (paddr 0x0081877f0) that resides in the TLB. 13 ROB entries later there is a load (LD1) from vaddr 0xcefeefe008 that has the store dependency mask bit set for the mentioned store. At the time LD1 is an incoming load in the LSU the store has not committed yet. The condition 

` val can_fire_load_incoming = widthMap(w => exe_req(w).valid && exe_req(w).bits.uop.ctrl.is_load)` 

is fulfilled and LD1 is sent to the TLB. Because of the prior invalidation of the corresponding TLB entry, LD1 misses and the PTW steps in and issues two hella_requests. The second request goes to address 0x00818777f0 (same as ST1). The cycle after that ST1 goes through to the DMEM. This RAW hazard leads to an invalid TLB entry that causes an exception and wrong address translation. Finally resulting in the core being stuck in a WFI. 

At this point I have only tested it with one Linux image. But the steps are:

1. 'make run-binary CONFIG=MediumBoomConfig BINARY=LINUX_IMAGE_NODISK'
2. wait 

I set the impact to unknown because there are a few ways to mitigate this. 
Shouldn't there be a sfence after setting up the page correctly, i.e., ST1? 
Another way around this would be to use the store dependency mask to check if a load can go to TLB or a probably more performant way would be to propagate the dependency masks to the hella requests and then preventing hella requests from executing. The latter would have a higher impact on the RTL due to the current "decoupled" handling of hella requests. 

[image]

Please let me know if I should provide more detailed information on anything. 

Kind regards

**What is the current behavior?**
MediumBoomConfig freezes during the boot process in simulation and on the FPGA.

**What is the expected behavior?**
MediumBoomConfig is able to boot in simulation and on the FPGA.

**Please tell us about your environment:**
    - version: `1ad28828`
    - build environment: `ucb-bar/chipyard` `9443466f`
    - but also tested with the most recent versions (chipyard 1.8.1 and the corresponding boom hash '[fac2c37](https://github.com/riscv-boom/riscv-boom/tree/fac2c370c9deae97ca52aca6b34857e9ac0f6e9d)'

jerryz123, 2022-12-19:
This was a bug in the Linux kernel. It was fixed some time ago, but we were slow to bump kernel versions in Chipyard.
You can try bumping to a newer version (as this PR does) https://github.com/firesim/FireMarshal/pull/257

Schmitz48, 2022-12-20:
Thank you! Now it works

closed 2022-12-20 by Schmitz48