riscv-boom/riscv-boom #636: Unable to read/write CSRx registers

avthakore, 2023-04-27:
**Type of issue**: bug report | feature request | question | other enhancement
Question

**Impact**: no rtl change | rtl refactoring | new rtl | unknown
No RTL Change

**Development Phase**: request | proposal

**Other information**

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

I am trying to write code for enabling and reading the performance counters in BOOM. My current setup is the LargeBOOM config for the boom core, running on the vcu 118 FPGA which is also running the bare metal flat Linux image. 

Whenever I try to read/write to the CSRx registers I get the following error.

`[15504.131895] testVersion2[328]: unhandled signal 4 code 0x1 at 0x0000000000010600 in testVersion2[10000+59000]
[15504.141836] CPU: 0 PID: 328 Comm: testVersion2 Not tainted 5.7.0-rc3-58539-g5f5fd87b36e2 #1
[15504.150264] epc: 0000000000010600 ra : 0000000000010706 sp : 0000003ffff27bc0
[15504.157362]  gp : 000000000006e9d0 tp : 0000000000074760 t0 : 0000000000000002
[15504.164679]  t1 : 2f2f2f2f2f2f2f2f t2 : 0000000000000001 s0 : 0000003ffff27c00
[15504.171979]  s1 : 0000000000000001 a0 : 0000000000000001 a1 : 0000003ffff27da8
[15504.179291]  a2 : 0000003ffff27db8 a3 : 0000000000000000 a4 : 0000003ffff27c28
[15504.186601]  a5 : ffffffffffffffff a6 : 000000000006ea80 a7 : 127d6a7c7a2f1d41
[15504.193903]  s2 : 0000003ffff27da8 s3 : 0000000000000001 s4 : 0000003ffff27db8
[15504.201217]  s5 : 0000000000000001 s6 : 00000000000105f6 s7 : 00000000000103e0
[15504.208530]  s8 : 0000000000000000 s9 : 0000002ac79e0390 s10: 0000002ac79dc630
[15504.215836]  s11: 0000000000000001 t3 : ffffffffffffffff t4 : 000000000006ca00
[15504.223140]  t5 : 0000000000073678 t6 : 0000000000069488
[15504.228545] status: 0000000200004020 badaddr: 0000000030679073 cause: 0000000000000002
Illegal instruction
`
Looking at the bad address, it is always some csr command. Relevent part of the code dump is attached 
` 105f6:	7139                	add	sp,sp,-64
   105f8:	fc06                	sd	ra,56(sp)
   105fa:	f822                	sd	s0,48(sp)
   105fc:	0080                	add	s0,sp,64
   105fe:	57fd                	li	a5,-1
   10600:	30679073          	csrw	mcounteren,a5
   10604:	57fd                	li	a5,-1
   10606:	10679073          	csrw	scounteren,a5
   1060a:	10100793          	li	a5,257
   1060e:	32379073          	csrw	mhpmevent3,a5
   10612:	6785                	lui	a5,0x1
   10614:	8017879b          	addw	a5,a5,-2047 # 801 <__libc_dlerror_result+0x7b1>
   10618:	32479073          	csrw	mhpmevent4,a5
   1061c:	fe042623          	sw	zero,-20(s0)
   10620:	a005                	j	10640 <main+0x4a>
   10622:	fec42783          	lw	a5,-20(s0)
   10626:	2785                	addw	a5,a5,1
   10628:	fcf42623          	sw	a5,-52(s0)
   1062c:	fec42783          	lw	a5,-20(s0)
`

Here I am trying to set the mcounteren for the super user where the program fails. 

I have also tried using the riscv code constructor suggested in past HPM issues and my error still persists. <link> https://github.com/abejgonzalez/riscv-code-constructor

I will be happy to share any other information required to debug this issue! 

Thank you team!

garthlei, 2023-04-28:
The mcounteren register should be accessed in machine mode. Where did you insert the code?

avthakore, 2023-04-28:
Hello! 

Thank you for the prompt reply. I am running this code as a statically compiled binary of the bare metal linux for VCU 118 FPGA. 

I am unsure as to how I can run a specific part of the code in Machine mode? Could you please share some links/directions?

garthlei, 2023-04-29:
I put the code in the BootROM. (Probably not a good practice but it works)

avthakore, 2023-05-03:
Hi! 
So I have been trying to figure out how to place my code into the bootrom. I see that I can create my own bootrom and then I have to make the FPGA bitstream again. That makes it impossible to quickly reiterate code. Am I looking at this the right way or is there some other way as well?

garthlei, 2023-05-07:
Inserting the code into OpenSBI (if you are using that) or the Linux kernel can also be an option, I guess; however, I am not familiar with either.