riscv-boom/riscv-boom #659: Dual core jammed while running unixbench

ARF1939261764, 2023-09-26:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**Other information**

**What is the current behavior?**
I tried to run unixbench using boom dual-core, but during the run, the cpu suddenly jammed, there was no error message, and the terminal did not respond, but the same code worked fine on rocket 4 core
**What is the expected behavior?**
Run unixbench with dual cores without jamming
**Please tell us about your environment:**

git hash:1a153d4974977a2d5f8baa9b4de51b3337d230fe
**What is the use case for changing the behavior?**

jerryz123, 2023-09-26:
What BOOM configuration did you use?

crsullivan13, 2024-05-08:
Not sure if I should open a new issue for this, but I am experiencing a similar problem. I am simulating a dual core, large boom configuration with a 1MB 2 bank LLC. All of this being simulated on FireSim.

When trying to co-run two of the Disparity workloads from the SD-VBS suite, the simulation hangs. I am also able to reproduce the issue with co-running a synthetic sequential write task that has LLC fitting working set.

I have opened a discussion on the Chipyard group as well.
https://groups.google.com/g/chipyard/c/rOhedyr7O-4