riscv-boom/riscv-boom #660: ECC support for BOOM I Cache and Cache

Kevin99214, 2023-10-04:
**Type of issue**: feature request 

**Impact**: new rtl / unknown

**Development Phase**: request 

**What is the use case for changing the behavior?**
In reference to this [issue](https://github.com/riscv-boom/riscv-boom/issues/579) and the [riscv-platform-spec](https://github.com/riscv/riscv-platform-specs/blob/main/riscv-platform-spec.adoc), currently rocket's ICache and DCache supports SECDED-ECC whereas BOOM's ICache and DCache does not. 
Is there any chance ECC will be added to BOOM ICache and DCache?

Kevin99214, 2024-01-18:
Closing this as it seems NBDCache does not support ECC. (#363 and chipsalliance/rocket-chip#495)

closed 2024-01-18 by Kevin99214

jerryz123, 2024-01-18:
BOOM does not use NBDCache anymore. Adding ECC to the self-contained caches would be an interesting project.