riscv-boom/riscv-boom #662: Build error

GGKOP, 2023-10-17:
I want to build boomV3 on the chipyard.
when I fellowed these  steps, I met some problems.
``````
# Download the template and setup environment
git clone https://github.com/ucb-bar/chipyard.git
cd chipyard
./scripts/init-submodules-no-riscv-tools.sh

# build the toolchain
./scripts/build-toolchains.sh riscv-tools

# add RISCV to env, update PATH and LD_LIBRARY_PATH env vars
# note: env.sh generated by build-toolchains.sh
source env.sh

cd sims/verilator
make CONFIG=LargeBoomConfig
``````` 
1.  the second step ::   bash: ./scripts/build-toolchains.sh: no such file or dir
what can I do ?
 
AMD® Ryzen 9 7950x 16-core processor × 32 Ubuntu 22.04.3 LTS 64bit

jerryz123, 2023-10-17:
These instructions are out of date. Can you try following the ones on the main Chipyard documentation site?

GGKOP, 2023-10-17:
[ latest chipyard doc site](https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html#initial-repository-setup)  thank you！