riscv-boom/riscv-boom #682: RV32IF/IMF Support Request

pulkitgupta1217, 2024-03-11:
**Type of issue**:  feature request | question

**Impact**: rtl refactoring | new rtl | unknown

**Development Phase**: request

BOOM currently does not support a 32-bit FPU and thus synthesis for RV32IMF or RV32GC, I saw that support for RV32 was added in #166 but #164 remains unresolved (and abandoned?) since the fpu support is not yet present.

Per the code/commits in #166, support for building boom with xlen and flen 32 was added [here](https://github.com/riscv-boom/riscv-boom/pull/166/files#diff-b165eed7db75bc52cdea7bc75fb937000cb45199a499df5f17f61095156a95ffR58), however using the 32 bit fpu is prohibited [here](https://github.com/riscv-boom/riscv-boom/pull/166/files#diff-05727df609bbd277c6106d1fc738d57e5865158f6554c1e3f85d5f718983147cR130).

In the current master branch, support for rv32 appears to have been dropped entirely, regardless of float support:
[WithBOOMRV32 config (or equivalent) no longer present](https://github.com/riscv-boom/riscv-boom/blob/a82f329f71fff2dc3cf08a3fb8e24bd71ff14bfa/src/main/scala/v4/common/config-mixins.scala)
[still checks for and prohibits rv32IF](https://github.com/riscv-boom/riscv-boom/blob/a82f329f71fff2dc3cf08a3fb8e24bd71ff14bfa/src/main/scala/v4/common/parameters.scala#L230)

Peeking around the code, it seems there are `flen == 64` requirements in many places and some commented out code for `xlen == 32` and `flen == 32`

**Question/Request**
Is it possible to add support for RV32GC in BOOM?

**What is the use case for changing the behavior?**
I believe there are applications for 32 bit out of order softcores and there may be a regression in support for RV32I

jerryz123, 2024-03-11:
RV32 OOO implementations may have some niche use cases, but RV64 OOO implementations are by far more useful. We decided early on that drop RV32 support in favor of more aggressive implementations.

A RV32 implementation could be recovered, but would require some substantial generator modifications. I'd be happy to review PRs implementing this.

pulkitgupta1217, 2024-03-11:
Unfortunately, implementing this myself would be far outside the scope of my PhD work.

Are there any well known rv32 ooo implementations with significant levels of parametrizability like what boom has?