riscv-boom/riscv-boom #687: Illegal Instruction Error on Accessing 'time' Register in S-mode Despite 'mcounteren' Set to Allow

youzi27, 2024-04-23:
**Type of issue**: bug report

**Impact**: unknown

**Description**
Hello,

I've encountered an issue with the BOOM processor where accessing the `time` register in S-mode results in `an illegal instruction exception`, even though the `mcounteren` register is set to `0x2` (enabling TIME bit). This behavior is inconsistent with both Spike's outputs and the RISC-V privilege specification.

**Expected Behavior** 
According to the RISC-V privileged specification section 3.1.11, setting the TM bit in the `mcounteren` register should allow reading the time register in S-mode without causing an illegal instruction exception.

**Actual Behavior** 
Despite setting the `mcounteren` to `0x2`, accessing the time register in S-mode results in an illegal instruction exception. However, other registers like `cycle` and `instret` **do not** exhibit this issue and can be accessed as expected.

**Reproduction Steps**
1. First set the TM bit of the mcounteren register in M mode:
```
addi t0, zero, 0x2
csrrw zero, mcounteren, t0
```
2. Then access it in S mode:
```
csrr t1, time
#
```
3. Check if it can be accessed normally without triggering any illegal instruction exceptions.

**Reference**
From the RISC-V privileged specification 3.1.11:

> When the CY, TM, IR, or HPMn bit in the mcounteren register is clear, attempts to read the cycle, time, instret, or hpmcountern register while executing in S-mode or U-mode will cause an illegal instruction exception. **When one of these bits is set, access to the corresponding register is permitted** in the next implemented privilege mode (S-mode if implemented, otherwise U-mode).

cyyself, 2024-04-23:
Just note that mtimecmp can only exposed in CLINT through MMIO and traps to M-Mode SBI to emulate the result. It’s not necessary to expose it as a regular CSR. Please read the specification carefully.

cyyself, 2024-04-23:
I wrote down the last comment when I was having lunch using my phone. Now I can tell you that you can find the reference in the spec about what I said here: https://github.com/riscv/riscv-isa-manual/blob/d8e41f9af1259bef31369383c356c69cefcdea83/src/machine.adoc?plain=1#L1496

> Implementations can convert reads of the `time` and `timeh` CSRs into
> loads to the memory-mapped `mtime` register, or emulate this
> functionality on behalf of less-privileged modes in M-mode software.

Please check the ISA spec first. This is not the first time I have seen you report some bug just for another implementation of the allowed behavior of RISC-V ISA. There are many design spaces in RISC-V, which is specification allowed.

youzi27, 2024-04-23:
Hi @cyyself ,

Thank you for your description of the issue. It's very helpful to me. If you find any issues with the problem I raised, feel free to comment. I appreciate your advice, and as a beginner in RISC-V, there are indeed many aspects allowed by the specifications that I need to carefully study. 

Thank you very much!

closed 2024-04-23 by youzi27